matth-x/MicroOcpp #225: Clock Aligned Meter values request

priya2212, 2023-10-28:
Hi @matth-x , 
I want to send meter values at every 30 min when there is no active transaction and during transaction, want to send meter values every 1 minute. 
I checked that we can do it with clock aligned data and sampled data respectively. But when I configure clock aligned data remotely through server, then charge point is sending the **sample.clock and sample.periodic** - both data during active transaction only.
Not sure, If I am understanding and implementing it correctly. Would really appreciate your valuable inputs on it.  

Thank you
Priya.

matth-x, 2023-10-30:
Hi @priya2212,

Sending MeterValues outside of transaction is good discussion material. See here for example some feedback: https://github.com/matth-x/MicroOcpp/issues/66

In case your backend tolerates MVs while there's no charging session, you could try changing the config `Cst_MeterValuesInTxOnly` to false. The charger will send some superfluous data (most readings are irrelevant while not charging), but I would argue that this is really little payload (probably too little to care about) and the true bottleneck of WebSocket-based charging networks is more the number of simultaneous open connections, where reducing traffic doesn't really help. Anyway, if the backend supports this behavior, then it should work as expected.

If that doesn't work, would it be an option to duplicate the energy meter inputs for connectorId 0? (i.e. add the same callback, but pass the parameter 0 for connectorId to `addMeterValueInput(...)`) That should send clock-aligned meterings regardless of open transactions.

priya2212, 2023-11-01:
So Sorry I got busy in some work here. Will check it today and confirm it here.

priya2212, 2023-11-02:
Dear @matth-x ,
Setting the Cst_MeterValuesInTxOnly to false works. Thank you very much for your support and help.

priya2212, 2024-01-27:
Dear @matth-x , 
I am configuring connector 0 to get the meter values irrespective of transactions. I have configured ClockAlignedDataInterval to 600 sec and MeterValueSampleInterval to 30 sec (to get frequent meter data when TX is there)
Expected output - When no tx is active, the charge point should only send clock-align data, and when tx is active - it should send sample data and send clock-align data. 

Actual output - When no tx is active, it is sending clock-align and sample data both. 
Because of this, I am having request queue highly populated and when I am trying to send RFID data for authorization, sometimes facing operation timeout issue. 

Do you think

Kundanjhakj, 2024-07-22:
Hi @priya2212,

In my current development, I need to start charging and stop charging when tapping the RFID. Can you please see if the steps I am following are correct or if any changes are required?

Steps I am following:

Start OCPP.
Send the meter value first.
Then send ocpp_beginTransaction and ocpp_endTransaction.

```
static void rfid_task(void *arg)
{
    while (1) {
        // Lock the mutex before reading rfid_current_value
        xSemaphoreTake(rfid_mutex, portMAX_DELAY);
        if ((rfid_current_value[0] != '\0') && (strlen(rfid_current_value) > 2)) {
            ESP_LOGI(TAG, "RFID: %s", rfid_current_value);

            if (transactionActive) {
                if (isSameTag(currentTag, rfid_current_value)) {
                    stopTransaction(rfid_current_value);
                }
            } else {
                strcpy(currentTag, rfid_current_value);
                startTransaction(rfid_current_value);
            }

            rfid_current_value[0] = '\0'; // Clear the RFID value after use
        }
        xSemaphoreGive(rfid_mutex);
        vTaskDelay(10 / portTICK_PERIOD_MS); // Yield to prevent watchdog resets
    }
}

```
```
void startTransaction(const char *rfid) {
    transactionActive = true;
    ESP_LOGI(TAG, "Starting transaction with tag: %s", rfid);
    ocpp_beginTransaction(rfid);
    // Add additional logic to handle transaction start if needed
}

void stopTransaction(const char *rfid) {
    transactionActive = false;
    ESP_LOGI(TAG, "Stopping transaction with tag: %s", rfid);
    ocpp_endTransaction(rfid, "EVDisconnected");
    memset(currentTag, 0, RFID_UID_LENGTH); // Clear the current tag
    // Add additional logic to handle transaction stop if needed
}

bool isSameTag(const char *uid1, const char *uid2) {
    return strcmp(uid1, uid2) == 0;
}
```

```
static void ocpp_task(void *arg)
{
    struct mg_mgr mgr;
    mg_mgr_init(&mgr);
    mg_log_set(MG_LL_DEBUG);

    struct OCPP_FilesystemOpt fsopt = { .use = true, .mount = true, .formatFsOnFail = true};
    OCPP_Connection *osock = ocpp_makeConnection(&mgr, EXAMPLE_MO_OCPP_BACKEND, EXAMPLE_MO_CHARGEBOXID, EXAMPLE_MO_AUTHORIZATIONKEY, "", fsopt);
    ocpp_initialize(osock, "ESP-IDF charger", "Tirex", fsopt, false);

    ocpp_setEnergyMeterInput(getEnergyReading);

    while (1) {
        mg_mgr_poll(&mgr, 10);
        ocpp_loop();
        vTaskDelay(100 / portTICK_PERIOD_MS); // Yield to prevent watchdog resets
    }

    ocpp_deinitialize();
    ocpp_deinitConnection(osock);
    mg_mgr_free(&mgr);
}
```
```
void app_main(void)
{
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
    init_UART();

    // Initialize the mutex
    rfid_mutex = xSemaphoreCreateMutex();
    if (rfid_mutex == NULL) {
        ESP_LOGE(TAG, "Failed to create RFID mutex");
        return;
    }

    xTaskCreate(echo_task, "uart_echo_task", ECHO_TASK_STACK_SIZE, NULL, 11, NULL); // Reading rfid value from another board
    xTaskCreate(ocpp_task, "ocpp_task", ECHO_TASK_STACK_SIZE, NULL, 10, NULL); // Increased stack size
    xTaskCreate(rfid_task, "rfid_task", ECHO_TASK_STACK_SIZE, NULL, 9, NULL); // RFID task with a lower priority
}
```