matth-x/MicroOcpp #226: OCPP server connection issue

MomentaryGlitch, 2023-10-30:
Hi,

Having an issue as below, using the TLS example as the is required for the OCPP server I have access to (vCharM)

I note that there is an OCPP_HOST unsure what this should be as I only have a URL for our server

[main] Wait for WiFi . connected
[main] Wait for NTP time (used for certificate validation) .. finished. Unix timestamp is 1698625898
[MO] info (MicroOcpp.cpp:296): initialized MicroOcpp v1.0.0
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected

matth-x, 2023-10-30:
Hi @MomentaryGlitch,

Given the OCPP connection info is like

Backend URL: `wss://example.com:8443/basepath`
ChargeBoxId: `charger001`

Then host is `example.com`, and URL is `wss://example.com:8443/basepath/charger001`

Meanwhile the ["non-TLS" example](https://github.com/matth-x/MicroOcpp/tree/master/examples/ESP) has also got TLS-capability and maybe it provides a nicer starting point. In case you need to provide an authentication key, fill the mocpp_initialize function appropriately.

MomentaryGlitch, 2023-11-01:
@matth-x Thanks for your response,

I did get the TLS example working, as per your above suggestion I have moved to the "non-TLS" example.

I have populated the mocpp_inialize function 

mocpp_initialize(OCPP_BACKEND_URL, OCPP_CHARGE_BOX_ID "My Charging Station", "My company name" password, CA_CERT);

#define password ""

Added required CA_Cert

const char CA_CERT[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
....
-----END CERTIFICATE-----
)EOF";

All built and loaded successfully but now I get the below response in the terminal -
[MO] ERROR (BootNotification.cpp:27): payload undefined

Can you please offer some feedback or possibly be able to elaborate on filling the mocpp_initialize function appropriately.

Also for your reference vCharM has now been pretty well tested with MicroOCPP so I would say compatible -

matth-x, 2023-11-01:
Thanks for the report that vCharM works, that's very useful!

Here's a full example of the `mocpp_initialize` function:

```cpp
const char *CA_CERT = "-----BEGIN CERTIFICATE----- ...";

mocpp_initialize(
    "wss://example.com:8443/basepath", //Backend URL
    "charger001",                      //Charge box ID
    "My Charging Station",             //Charge point model
    "My company name",                 //Charge point vendor
    MicroOcpp::FilesystemOpt::Use_Mount_FormatOnFail,    //Flash strategy: let OCPP lib mount flash and format if needed
    "websocket_pass_123",              //WebSocket authentication key
    CA_CERT,                           //TLS certificate
    false);                            //Disable auto-recovery of local transaction cache (recommended for development)
```

MomentaryGlitch, 2023-11-01:
@matth-x 

Worked a charm. Thanks for your help!

closed 2023-11-01 by MomentaryGlitch