matth-x/MicroOcpp #227: Transaction status after PowerLoss

NuriaParra8997, 2023-10-31:
Hello everyone.
I have noticed that if my device runs out of power during a charging session, after regaining power (reboot), the functions ocpp_isTransactionActive and ocpp_isTransactionRunning return negative.
However, in Steve, the session remains active.
I have tried to cancel this reservation with ocpp_endTransaction(NULL, "PowerLoss"), but it doesn't work for me.

I would like to know if there is an active transaction so that I can continue supplying power to the vehicle.

Thank you very much!

matth-x, 2023-10-31:
Hi @NuriaParra8997,

Sounds to me like the filesystem access doesn't work. Do you see any suspicious log messages related to this? Probably it will be helpful to increase the log level from `DL_INFO` to `DL_DEBUG`.

Navjot0005, 2023-11-20:
@NuriaParra8997 You can try this format because it works with me.
endTransaction("CARD number if any","PowerLoss",1U);

closed 2024-01-09 by NuriaParra8997