matth-x/MicroOcpp #228: Can we run this in Linux OS - Ubunutu?

GanesanGuru, 2023-11-01:
Hi @matth-x ,
Is it possible to run this codebase directly from Linux OS - Ubunutu in VCode?
I'm geeing unknown error in terms of Arduino.h or ESP8266WiFi.h files? How to rectify those @matth-x ?

PS : SteVe OCPP Server has been setuped successfully..

matth-x, 2023-11-01:
Hi @GanesanGuru,

Yes, this library runs on Ubuntu too. A full application reference is the [MicroOcppSimulator](https://github.com/matth-x/MicroOcppSimulator) which includes all necessary settings.

GanesanGuru, 2023-11-02:
Hi @matth-x ,
I've installed [MicroOcppSimulator](https://github.com/matth-x/MicroOcppSimulator) and I can run it. But not sure how to workaround with it. Does it has a inbuilt ocpp server? can we connect to external ocpp-server like Steve? Is there any step-by-step blog/manual to understand it? like

1. This acting as client only; which connects to external Steve like ocpp server?
2. How to execute all the transactions from client to server & vice-versa too? in terms of step-by-step?

GanesanGuru, 2023-11-02:
Also @matth-x ,
I've downloaded the [ArduinoJson](https://github.com/bblanchon/ArduinoJson) but not sure how to include that into microOCPP, as lot of files have <ArduinoJson.h> ?

GanesanGuru, 2023-11-02:
Any suggestions @matth-x

priya2212, 2023-11-04:
@GanesanGuru  for the first question, I believe It doesn't have any OCPP server, it is just a OCPP-client which by default connects to ws://echo.websocket.events/ server with charge box id- charger-01. We can change the OCPP server URL when our client is up, then using web interface - in the OCPP 1.6 Websocket tab, we can update the URL and charge box ID with our server and ID. 

We can start and stop the transactions through our server or steve server accordingly and can see the status changes in the logs at both and server end.

GanesanGuru, 2023-11-06:
Hi @matth-x @priya2212  ,
I've done as mentioned by @matth-x  ; created Platform IO and incorporated the project and tried to build the example main.c within ESP. Yet, there is an error as attached below.
[image]
In the attachment, check the platform is nodemcu only & I've used build icon to build it and its corresponding console is shown. Where did I it went wrong?

Do I need to pass the compilation command with flags manually via terminal? If so, do let me know asap. Sorry for bothering too...

closed 2023-11-07 by matth-x