matth-x/MicroOcpp #229: Stepwise guidance in implementing in ESP32/Ubuntu?

GanesanGuru, 2023-11-02:
Hi @matth-x  @all,
Is there anyone has successfully incorporated this library into ESP32 and if so kindly aid the knowledge that has been acquired during the process. There are some you could aid me/newbie like me attain the same too.

The stepwise guidance for ESP32 / any Linux based OS is very welcomed.

PS : As @matth-x being very busy, I'm raising this concern more to the community. Setup of ocpp-server Steve is attained. Hence aid us in implementing the client side part alone. Thanks in advance.

matth-x, 2023-11-03:
Hi @GanesanGuru,

Yes, I must admit that the documentation of this open source project is by far not as extensive as it should be. However, there is some material. A [technical introduction](https://matth-x.github.io/MicroOcpp/intro-tech/) explains the basic principles of this library. The API documentation can be found in [MicroOcpp.h](https://github.com/matth-x/MicroOcpp/blob/master/src/MicroOcpp.h).

I think that most people get started with this library by creating an example project in PlatformIO based on the [ESP example](https://github.com/matth-x/MicroOcpp/tree/master/examples/ESP). When you manage to build this project successfully in PlatformIO, then you have a good reference project and the perfect resource to learn how including MicroOcpp generally works. The corresponding starting point on Ubuntu would be to clone and build the [Simulator](https://github.com/matth-x/MicroOcppSimulator).

GanesanGuru, 2023-11-06:
Hi @matth-x ,
The corresponding starting point on Ubuntu would be to clone and build the [Simulator](https://github.com/matth-x/MicroOcppSimulator).

This line has saved me from all that uncertain EVSE charging environment. Yes, I was able to successfully integrate the SteVe with MicroOCPP Simulator and able to perform transactions,etc and all. I've few queries relate to OCPP 1.6J spec & MicroOCPP Simulator implementation, as listed below:

1. I was able to re-create 6/9 states of EV Charger - while I'm unable to recreate - Reserved,Unavailable ,Faulted states
2. I was able to re-create 7/10 of ChargePoint Transactions - while I'm unable to recreate - Data Transfer,Diagnostics & Firmware Status Notification
3. I was not able to recreate any of total 19 - Operations Initiated by Central System - like reservation,stop/start transactions using SteVe...
4. Whether microocpp has impleted all those leftout operations &whether because of SteVe I'm not able to visualize it? If so, recommended any OCPP server that you have used to develop the simulator...