matth-x/MicroOcpp #230: Charger status Unavailable and timestamp at default time 1970s

chandan-ultraviolette, 2023-11-03:
Hello, OCPP sending status packet as below. I know I have to set the status and timestamp but need the function name for this. Any is there any quick guide on what all function should be set when integrating OCPP with main application.

[2,"1000003","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Unavailable","timestamp":"1970-01-01T00:01:10.932Z"}]

Thanks

chandan-ultraviolette, 2023-11-03:
Or once connected with CMS, CMS will update these parameters ?

Also MOCPP_NUMCONNECTORS=2  means there are 2 connector at charging station bcz below 2 param OCPP_ID_OF_CP and OCPP_ID_OF_CONNECTOR is confusing, can any explain ?

#define OCPP_ID_OF_CP 0
#define OCPP_ID_OF_CONNECTOR 1

matth-x, 2023-11-04:
Hi @chandan-ultraviolette,

I never had this (or a similar) error case so I can't share any ideas about this right away. Can you maybe look for further relevant log messages or other suspicious behavior? And did you change the config `Cst_PreBootTransactions` to true by any chance?

In any case, the first message this library sends (BootNotification) doesn't succeed. That's where the clock would have got set.

The build flag `MOCPP_NUMCONNECTORS` refers to the number of "logical connectors" as defined in the OCPP specs. The number of logical connectors is always 1 higher than the number of actual (physical) connectors. You can find more background about this in the OCPP specs.

chandan-ultraviolette, 2023-11-05:
Cst_PreBootTransactions is the function? I searched but didn't find any such keyword.

I can find this doc: https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pdf or any other document. Incase I want to have 2 charging guns means 2 connector, so just need to increase MOCPP_NUMCONNECTORS to 3 or need to do anything other then this ?

matth-x, 2023-11-07:
Then `Cst_PreBootTransactions` is not the issue. I would guess there went something wrong on the WebSockets side but more information would be required to actually tell you the problem here (i.e. original WebSocket messages, OCPP lib logs, SteVe logs).

> Incase I want to have 2 charging guns means 2 connector, so just need to increase MOCPP_NUMCONNECTORS to 3 or need to do anything other then this ?

Yes, exactly. Define this flag in the build settings of your project and it will overrule the default value.

closed 2023-11-09 by chandan-ultraviolette