matth-x/MicroOcpp #231: PhaseSwitching via SetChargingProfile - howto

Pablo2048, 2023-11-03:
Hi Matthias,
thank You very much for writing (and publishing) this library. I've successfully integrated (very old) version 0.0.1 and now I'm switching to version 1.0.0. My question is that I need to implement `ConnectorSwitch3to1PhaseSupported`, followed by used phase switching via SetChargingProfile. Do You have any guide, or how-to You can suggest please?

Thank You very much!

Pablo2048, 2023-11-04:
So the question is - I probably have to use https://github.com/matth-x/MicroOcpp/blob/8744bd00e92f7fac1de3c2b49c6fdce5af49a87b/src/MicroOcpp.h#L260 to set the correct callback and set `ConnectorSwitch3to1PhaseSupported` the same way as old `AO_FreeVendActive` (new `Cst_FreeVendActive`) flag via `MicroOcpp::declareConfiguration<bool>()`... The flag name would be probably `Cst_ConnectorSwitch3to1PhaseSupported`. Is all  this correct?

matth-x, 2023-11-04:
Yes, that's the best way to integrate this feature.

Since `ConnectorSwitch3to1PhaseSupported` is read-only, the more standard-conform way to add it would be

```cpp
//somewhere after mocpp_initialize(...)

MicroOcpp::declareConfiguration<bool>(
    "ConnectorSwitch3to1PhaseSupported", //configuration key
    true,                                //default value (= final value because this is read-only)
    CONFIGURATION_VOLATILE,              //optimization: don't store this config persistently on flash, but keep it in volatile memory (heap)
    true);                               //read-only flag as reported in GetConfiguration
```

Only the prefix `Cst_` is not needed here because `ConnectorSwitch3to1PhaseSupported` is part of the OCPP standard, in contrast to the custom `Cst_FreeVendActive` config.

And the `setSmartChargingOutput` you found is the correct one. The third parameter of the callback `chargingLimitOutput` is the number of phases.

Btw. happy about the update from your side!

Pablo2048, 2023-11-05:
Thank You very much for explanation - please don't close this, because I certainly write some info about progress and final info, when I'm finished with this task...

Pablo2048, 2023-11-29:
Ok, I have some news - I've successfully implemented `ConnectorSwitch3to1PhaseSupported` variable with right content and hooked up `setSmartChargingOutput` so everything looks good. Thanks again @matth-x for the library and guidance.

closed 2023-11-29 by Pablo2048