matth-x/MicroOcpp #235: ESP32:  [MO] info (Connection.cpp:65): Disconnected

ahmed-adel-hne, 2023-11-20:
Hi
I have a problem when conneting to the:
test tool the connection is successed and I can test the charger

but when I try to connect to the real backend server it show always:

[MO] info (Connection.cpp:65): Disconnected
[MO] info (RequestQueue.cpp:45): operation timeout: BootNotification

I use the normal example without TLS and the backend server not using TLS but they use port 8081 instead of 80 and I tried too many cases the output is:

[main] Wait for WiFi: ... connected!
[MO] info (MicroOcpp.cpp:131): connecting to ws://************.com:8081/OCPP/Test1234 -- (host: jadawelrealestate.com, port: 8081)
[MO] info (MicroOcpp.cpp:296): initialized MicroOcpp v1.0.0

ahmed-adel-hne, 2023-11-20:
[MO] info (MicroOcpp.cpp:296): initialized MicroOcpp v1.0.0
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected

feyzanoz, 2023-11-27:
I'm experiencing a similar issue. I'm trying to set up and test the MicroOCPP simulator and dashboard, but it's not communicating with the ESP32 board. If I leave it in its default state, I get the following output:
-----------------------------------------------------------
[MO] info (MicroOcpp.cpp:131): connecting to ws://echo.websocket.events/esp-charger -- (host: echo.websocket.events, port: 80)
E (5088) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1347:error: Corrupted dir pair at {0x0, 0x1}
E (5088) esp_littlefs: mount failed,  (-84)
E (5091) esp_littlefs: Failed to initialize LittleFS
[MO] info (MicroOcpp.cpp:296): initialized MicroOcpp v1.0.0
[MO] info (Connection.cpp:68): Connected to url: ws://echo.websocket.events/esp-charger
[MO] warning (RequestQueue.cpp:182): Invalid input! Not a JSON
[MO] warning (Connection.cpp:76): Processing WebSocket input event failed
[MO] info (BootNotification.cpp:83): request has been Accepted
[MO] info (StatusNotification.cpp:50): New status: Available (connectorId 0)
[MO] info (StatusNotification.cpp:50): New status: Available (connectorId 1)
[MO] info (SmartChargingService.cpp:513): New limit for connector 1, scheduled at = 2022-01-28T11:59:55.012Z, nextChange = 2037-01-01T00:00:00.000Z, limit = {-1.000000,-1.000000,-1}
-----------------------------------------------------------
I receive this output. If I connect to port 8000, I see the same "disconnect" message.

ahmed-adel-hne, 2023-11-27:
I solved this problem by configure the url manually in the webSocket.begin()
 in the MicroOcpp.cpp file to

/dir/charger 

instead of ws:// ......... /dir/charger

make sure that the port is valid for OCPP websocket connection

feyzanoz, 2023-11-28:
Thank you for your response. There was an issue with the server where my connection address is hosted. I have resolved the issue.

ahmed-adel-hne, 2023-11-28:
@matth-x
Yes it shows as you typed and it also disconnected
when I modified the library to parse the url and remove the first part just keep /dir/charger and path this new url to the websocket.begin() it worked perfectly

ahmed-adel-hne, 2023-11-28:
Thank you very much for your effort you helped me so much.

closed 2023-12-02 by ahmed-adel-hne