matth-x/MicroOcpp #237: startTransaction API will generate 2 "StartTransaction" mesages

polliwog, 2023-12-17:
"OCPP1.6" TX:  "[2,\"1000005\",\"StartTransaction\",{\"connectorId\":1,\"idTag\":\"0000001012951691\",\"meterStart\":1,\"timestamp\":\"2023-12-17T07:48:40.564Z\"}]"
"OCPP1.6" RX:  "[3,\"1000005\",{\"transactionId\":113,\"idTagInfo\":{\"status\":\"Accepted\",\"expiryDate\":\"2030-02-03T00:00:00.000Z\"}}]"
"OCPP1.6" TX:  "[2,\"1000006\",\"StartTransaction\",{\"connectorId\":1,\"idTag\":\"0000001012951691\",\"meterStart\":1,\"timestamp\":\"2023-12-17T07:48:40.564Z\"}]"
"OCPP1.6" RX:  "[3,\"1000006\",{\"transactionId\":113,\"idTagInfo\":{\"status\":\"ConcurrentTx\",\"expiryDate\":\"2030-02-03T00:00:00.000Z\"}}]"
"OCPP1.6" TX:  "[2,\"1000007\",\"StatusNotification\",{\"connectorId\":1,\"errorCode\":\"NoError\",\"status\":\"Charging\",\"timestamp\":\"2023-12-17T07:48:40.566Z\"}]"
"OCPP1.6" RX:  "[3,\"1000007\",{}]"
"OCPP1.6" TX:  "[2,\"1000008\",\"StopTransaction\",{\"meterStop\":1,\"timestamp\":\"2023-12-17T07:48:40.695Z\",\"transactionId\":113,\"reason\":\"DeAuthorized\"}]"
"OCPP1.6" RX:  "[3,\"1000008\",{}]"
"OCPP1.6" TX:  "[2,\"1000009\",\"StatusNotification\",{\"connectorId\":1,\"errorCode\":\"NoError\",\"status\":\"Finishing\",\"timestamp\":\"2023-12-17T07:48:40.698Z\"}]"

matth-x, 2023-12-17:
The logs show that the StartTransaction operations have a different messageId, i.e. they are two distinct operations. Could it be that somehow `startTransaction(...)` gets called twice?

In case you don't need precise control over when messages are sent, I would recommend using `beginTransaction(...)` and `endTransaction(...)` instead of start- and stopTx. They avoid bad message flows as shown in these logs, and facilitate offline transactions and keeping transactions in sync over blackouts. See the documentation in MicroOcpp.h for more info.

polliwog, 2023-12-24:
If I use beginTransaction(...), how to get the server's return? (startTransaction(...)  have callback, like OnReceiveConfListener)