matth-x/MicroOcpp #238: TinyGSM instead of Wi-FI

ziyadtarek7, 2023-12-17:
Hello @matth-x 

Do you have any idea if is it possible to use TinyGsmClient instead of WifiClient? I'm trying to connect to the OCPP server over sim800l but I'm not able to find where I can should replace client object.

If you can point to where should I start, I would appreciate it so much.
Thank you

matth-x, 2023-12-19:
Hi @ziyadtarek7,

I believe that TinyGSM implements the Arduino Client interface. Since arduinoWebSockets is programmed against WiFiClient (also implementing Arduino Client) there's a good chance that you can replace all references to WiFiClient with TinyGSM.

There's been some discussions about this in this issue section. A second approach would be to look for another WebSockets library which already brings built-in support for the SIM800L modem and then to make an adapter around it that suits the MicroOcpp `Connection` interface. That approach works for sure, but is a little bit more work than trying to modify the existing WebSocket integration.

For your reference, here are other issues discussing this matter:
#188
#90 
#107 
#219

ziyadtarek7, 2023-12-19:
@matth-x 
I was able to get it working, now I will look into getting multiple network interfaces (wifi, ethernet and gsm) working simultaneously so that I'm able to switch to the most stable interface.

Should I share what I have modified to get it working here? In case anyone else faced the same issue

matth-x, 2023-12-20:
Oh cool!

It would be incredibly useful to share the code online, because 4G connectivity is discussed a lot here.

I assume you've forked arduinoWebSockets? The maintainer of that project is very open to contributions, so if you find the time to add the TinyGSM support there, then anybody can use it with MicroOcpp out of the box. But that needs time, just sharing the code here would already be great.

Avi241, 2023-12-31:
Hey @ziyadtarek7 . It was really great that you get it working even I was trying something similar with A7672S GSM 4G LTE module. It will be really helpful if you can share your work or guide me for the same. Thanks

ziyadtarek7, 2023-12-31:
> Hey @ziyadtarek7 . It was really great that you get it working even I was trying something similar with A7672S GSM 4G LTE module. It will be really helpful if you can share your work or guide me for the same. Thanks

Hello @Avi241, I will share all my work on this topic, hopefully, this weekend (Thursday or Friday)

That's what I was able to get working:
- OCPP over tinyGSM (Sim800)
- Switching between WiFi and GSM during runtime

I had a problem with getting secure websockets working on tinyGSM ( using SSL) so for now I have removed this part

ziyadtarek7, 2024-01-04:
Hello, 
This my modifications that got MicroOCPP working with SIM800 module instead of WiFi

https://github.com/ziyadtarek7/Using-MicroOCPP-with-SIM800-Module/tree/main

Avi241, 2024-01-20:
Thank you so much @ziyadtarek7 . I will look into it and let you know

Avi241, 2024-02-06:
This really worked out of the box @ziyadtarek7  . Thank you so much for the help. Also I am looking towards making TinyGSM to make it compatible with SIMCOM A7672s 4G GSM module as I want my system to be compatible with 4G network. If you have any idea how to achieve this it will be helpful or else even I will keep on posting my progress for 4G module.

ziyadtarek7, 2024-02-07:
> This really worked out of the box @ziyadtarek7  . Thank you so much for the help. Also I am looking towards making TinyGSM to make it compatible with SIMCOM A7672s 4G GSM module as I want my system [...]

Unfortunately I have no experience with that module, but if it has an AT commands interface maybe you could edit the commands sent from tinyGSM to match those of the module

SwapB5632, 2024-09-08:
@ziyadtarek7  did you make any changes in arduinoWebSockets or microOCPP for this

SwapB5632, 2024-09-08:
@ziyadtarek7 I'm getting this error

[image]

demirschmid61, 2024-09-18:
@ziyadtarek7 

Thank you very much for your repo, I downloaded and it worked perfectly **only for gsm**

Now I want to make it work for wifi as well, when I saw the websocket.h file from the repo, the client is commented, can you please throw some light on how we can use both wifi and tinygsm client.

ziyadtarek7, 2024-12-07:
> @ziyadtarek7
>
> Thank you very much for your repo, I downloaded and it worked perfectly **only for gsm**
>
> Now I want to make it work for wifi as well, when I saw the websocket.h file from the repo, [...]

I'm sorry, I didn't see your comment earlier.

I was able to get both wifi and tinygsm working and switching between them, I will look for the code and will share it. But here is what I did,

I wasn't able to switch between wifi and gsm during runtime because Arduino websockets library used macros to define which client to use so I created another version of the library that uses only gsm, then I modified microocpp lib to be able to use both the original version of Arduino websockets and the modified version,
Now what I have to do is just pass the object of whichever version I want to use when configuring microocpp