matth-x/MicroOcpp #240: crash due to appending of sampled values in  MeterValues response

Aylaa1, 2023-12-22:
the sampled meter values are appending in the Metervalues response every time the response is sent to the server.
after the Metervlaue size reaches 30k bytes as response, there is no meter value in the subsequent metervalue response,( [OCPP] Send: [2,"1000023","MeterValues",{"connectorId":1,"transactionId":1702437577,"meterValue":[{}]}]
[OCPP] Recv: [4,"1000023","InternalError","internal error",{}])
the code is crashed after this.
I have attached snips of log 
[image]

matth-x, 2023-12-28:
Hi @Aylaa1,

You need to set the MeterValueInput only once. The callback will be valid for the entire runtime of the library. When the MeterValueInput is set, then the library will fetch the metering data by calling that function.

The crash occurs when the OCPP server rejects the MeterValue message. Thanks for reporting this bug, I will reproduce it and fix it soon!

Aylaa1, 2024-01-11:
Hi @matth-x ,
Is there any update or release of the fixed version on this issue?

Thanks,
Aylaa