matth-x/MicroOcpp #241: crash in the file ocpp-config.jsn

Aylaa1, 2023-12-22:
Hi @matth-x,

I observed that the json capacity is going up to 1459 for the occp-config.jsn file, and due to this the code is crashed.
If I limit the size to 1024, the json file is invalid.
is there something I am missing in the filesystem?

Aylaa1, 2024-01-08:
Hi @matth-x , is there any update on this issue? could you please prioritize.
Thank you,
Aylaa

matth-x, 2024-01-08:
Hi @Aylaa1,

To change configurations programatically, use the Configuration API like here:

https://github.com/matth-x/MicroOcpp/blob/735efa29805d33723d80cd351e4ca773b0d96a3e/tests/Metering.cpp#L98-L99

The first call gets a handle for the Config, the second call effectively changes its value. The same functions exist for strings and bools.

Also `#include <MicroOcpp/Core/Configuration.h>` to get access to the Configuration API. Perform the changes after mocpp_initialize().

Manually manipulating the contents of the JSON files is without effect after initialization of the library. And before initialization it is risky, because I do not test such interaction with the firmware.

Aylaa1, 2024-01-10:
Hi @matth-x,
Thanks you for the response,
the sequence you have mentioned is followed the same way, i.e., mocpp_initialize() is called first and later the configuration related apis.
The actual problem is when the ocpp-config.json file is reaching the size above 1024(eg 1031), the application crashes.
If I restrict the jsonCapacity to 1024  in (DynamicJsonDocument doc {jsonCapacity}), but the jsonCapacity calculated in the runtime is 1031, json file would be invalid and the application still crashes.
The crash still persists if I increase the jsonCapacity to Max json capacity(4096).
Basically I observe crash if I try to write anything more than 1024 bytes to ocpp-config.jsn file.

The problem does not occur, if I reduce the number of key-value pairs written to the ocpp-config.jsn file in such a way that the file size will not cross 1024 bytes.

(All other jsn files are below 1024 bytes except ocpp-config.jsn file, throughout the application like bootstats.jsn, ws-conn.jsn, sc-tx-1-0.jsn, ...)

PFA, the crash scenario log snips
[image]
[image]

Thanks,
Aylaa

matth-x, 2024-01-14:
Hi @Aylaa1,

Sorry, I'm still lacking an idea what could happen here. The most suspicious thing I see is that the application crashes due to a stack overflow. Normally, the stack stack size should rarely grow with payload data. Did you integrate a custom filesystem adapter which has payload buffers on the stack by any chance?

However, it is true that the default stack size in esp-idf is too small to run MicroOcpp. I increased it to 8192 in the example project:

https://github.com/matth-x/MicroOcpp/blob/735efa29805d33723d80cd351e4ca773b0d96a3e/examples/ESP-IDF/sdkconfig#L505