matth-x/MicroOcpp #242: Firmware Update ESP32  ([MO] info (FirmwareService.cpp:160): Installation timeout or failed! Retry)

engahmedaadel, 2023-12-24:
Hi Everyone,
I have a problem when try to update the firmware, I use evse-test-tool to test the firmware update , any idea where is the problem?
I use github to download the .bin file

Here the log data:

[MO] info (FirmwareService.cpp:209): Scheduled FW update!
                  location = https://raw.githubusercontent.com/ahmed/main/main.bin
                  retrieveDate = 2023-12-24T17:04:37.160Z
                  retries = 1, retryInterval = 180
[MO] info (FirmwareService.cpp:60): Start update
[MO] info (StatusNotification.cpp:50): New status: Unavailable (connectorId 0)
[MO] info (FirmwareService.cpp:137): Installing
[112035][E][HTTPUpdate.cpp:234] handleUpdate(): HTTP error: connection lost

[MO] warning (FirmwareService.cpp:343): HTTP_UPDATE_FAILED Error (-5): HTTP error: connection lost

[MO] info (Connection.cpp:65): Disconnected
[MO] info (FirmwareService.cpp:160): Installation timeout or failed! Retry

Thanks.

matth-x, 2023-12-28:
Hi @engahmedaadel,

The issue here is that the ESP firmware updater doesn't support TLS, it works over HTTP only. Sorry that its limitations are everything but clear. I will either add a meaningful warning statement in future or declare it as example code only.

To try the firmware update procedure with the ESP updater I always ran the Python Simple HTTP server locally ([Howto](https://developer.mozilla.org/en-US/docs/Learn/Common_questions/Tools_and_setup/set_up_a_local_testing_server#using_python)) for hosting the binary. That did the job well.

matth-x, 2023-12-28:
Btw. I did a short code review and it seems like HTTPS support could be enabled by just adding two lines. 

This is the place in the OCPP lib to change manually:

https://github.com/matth-x/MicroOcpp/blob/735efa29805d33723d80cd351e4ca773b0d96a3e/src/MicroOcpp/Model/FirmwareManagement/FirmwareService.cpp#L334-L336

Here is what to add: https://github.com/espressif/arduino-esp32/blob/f6e12eb7e900fd6484e186b4b681f5b9504d2e85/libraries/HTTPUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino#L95-L96

engahmedaadel, 2023-12-28:
Thank you @matth-x  for your support, It mean a lot to me.
I'll try your solution and give you a feedback

demirschmid61, 2024-02-22:
Thank you @matth-x for your support,

I followed the steps for HTTPS firmware update, its working. The file is downloading and installing perfectly, but the CP is not able to send the following FirmwareStatusNotification,

1. **Downloading**
2. **Downloaded**

I can see only the installing status and installed when the charger reboots.

Is there anything, I need to add to the library?

```
[MO] Recv: [2,"2055a53f-1a2a-45c8-b968-fd0a62b61e2e","UpdateFirmware",{"location":"https://raw.githubusercontent.com/abcd/sdscsdc/firmware.bin","retrieveDate":"2024-02-22T17:11:06.260Z"}]
[MO] info (FirmwareService.cpp:211): Scheduled FW update!
                  location = https://raw.githubusercontent.com/demirschmid61/esp32update/main/firmware.bin
                  retrieveDate = 2024-02-22T17:11:06.260Z
                  retries = 1, retryInterval = 180
[MO] Send: [3,"2055a53f-1a2a-45c8-b968-fd0a62b61e2e",{}]
[MO] info (FirmwareService.cpp:62): Start update
[MO] debug (Connector.cpp:391): Status changed
[MO] info (StatusNotification.cpp:50): New status: Unavailable (connectorId 0)
[MO] Send: [2,"1000010","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Unavailable","timestamp":"2024-02-22T17:11:06.941Z"}]
[MO] Recv: [3,"1000010",{}]
[MO] Send: [2,"1000011","FirmwareStatusNotification",{"status":"**Installing**"}]
[MO] Recv: [3,"1000011",{}]
```