matth-x/MicroOcpp #243: Adding Smart Charging output limit functionality

priya2212, 2023-12-26:
Hi @matth-x , Hope you're doing great and Merry Christmas! :)
I am trying to implement Smart charging limit functionality. I am not able to fully understand it as not able to integrate it properly in my application. 
void ocpp_setSmartChargingOutput(OutputSmartCharging chargingLimitOutput);

What I am doing - 
creating a function with power, current, phases limit - 

```
void max_output_power_set(float power, float current, int nPhases)
{
    float voltage = 230.0;
    nPhases = 1;
    current = 32.0;
    int duty_cycle = 0;

    power = current * voltage * nPhases;

    ESP_LOGI(TAG, "max power delivery is %.2f ", power);

    if (current < 51.0)
    {
        duty_cycle = current / 0.6;
    }
    else
    {
        duty_cycle = (current / 2.5) + 64;
    }
    int duty_value = pow(2, LEDC_DUTY_RES) * (duty_cycle / 100);
    ESP_ERROR_CHECK(ledc_set_duty(LEDC_MODE, LEDC_CHANNEL, duty_value));
    ESP_ERROR_CHECK(ledc_update_duty(LEDC_MODE, LEDC_CHANNEL));
}
```

Calling it in main - 
`   ocpp_setSmartChargingOutput(max_output_power_set);
`

But running this program, I am getting limit as follows - 
[MO] info (SmartChargingService.cpp:509): New limit for connector 1, scheduled at = 2023-12-26T17:34:43.674Z, nextChange = 2037-01-01T00:00:00.000Z, limit = {-1.000000,-1.000000,-1}

I am not able to see the values I have set in the function. Am I implementing it wrong or understanding it wrong?
Would be really great if you can give me some of your valuable inputs.

matth-x, 2023-12-28:
Hi @priya2212, thank you!

The idea of the SmartChargingOutput is as follows: the OCPP server sends charging schedules to the OCPP lib, the lib stores and interprets them. While being in a charging session, the OCPP lib regularly checks the charging schedules if their charging limit changes at this point in time. If yes, then the OCPP lib notifies your code by calling the SmartChargingOutput (`max_output_power_set`) with the new values.

For example, lets assume the server sent a charging schedule which limits charging to 24 amps for the first hour of charging and 16 amps afterwards. Then your SmartChargingOutput (`max_output_power_set`) will be called two times by the OCPP lib:

- first time, when transaction begins: _power=-1 (no limit defined)_, **current=24**, _nphases=-1 (no limit defined)_
- second time, after 1 hour: _power=-1 (no limit defined)_, **current=16**, _nphases=-1 (no limit defined)_

The responsibility of the SmartChargingOutput is to take those values and forward them to the electronics of the charger (e.g. by updating the duty cycle of the PWM output like you do). Overriding these parameters will have no effect. If power / current / nphases is < 0, the values are not defined by a charging schedule (i.e. you can charge at the maximum power). If one or multiple of them are defined, you usually need a short logic of picking the "lowest" of all values ([see here](https://github.com/matth-x/ESP32_WiFi_V4.x/blob/e178698e3bd9f1fe107395b51da568bfb7d5bd2b/src/ocpp.cpp#L313-L327) for an example).

priya2212, 2023-12-28:
Hi @matth-x , 
Thank you for clear explanation. It clears my doubts now.

closed 2023-12-28 by priya2212