matth-x/MicroOcpp #244: Transaction still running after calling endTransaction(...)

pedro-fuoco, 2023-12-27:
Hi @matth-x, I'm back to work with MicroOcpp! 
It's great to see how the library progressed over the last semester and I'm ready to contribute again.

Currently I'm facing a particularly weird occurance... I've updated from `ArduinoOcpp` to `MicroOcpp` v1.0.0. And  I've tried to start a transaction, which went fine, but when I stopped the transaction and started it again (by reconnecting the evPlug, which makes the `isCharging()` function return `true`) the following appeared:

`[MO] warning (Connector.cpp:534): tx process still running. Please call endTransaction(...) before`

Naturally, I tried ending the transaction before calling `beginTransaction(...)` again, nothing changed...

I then followed your suggestion from #174 and added the following to the start of my code:
```cpp
endTransaction(nullptr, "PowerLoss");
```

Still, nothing changed. Using `stopTransaction(...)` reveals that no tx process is running, which contradicts the other log message.

`isTransactionRunning(...)` returns false but `getTransactionIdTag(...)` returns the tag I used in the last `beginTransaction(...)`, which seems weird. Any idea on what might be happening ?

The logic in the code is similar to this one, which worked fine in `ArduinoOcpp`:
```cpp
if (isCharging(connectorId) && !isTransactionRunning(connectorId))
    {
      Serial.println("[main] Start of transaction.");
      beginTransaction(idTag ,connectorId);
    }
    if (!isCharging(connectorId) && isTransactionRunning(connectorId))
    {
      endTransaction(idTag, "Local", connectorId);
    }
```

pedro-fuoco, 2023-12-27:
Here is an extended log:
```
[MO] info (BootNotification.cpp:83): request has been Accepted
[MO] info (StatusNotification.cpp:50): New status: Available (connectorId 0)
[MO] info (StatusNotification.cpp:50): New status: Preparing (connectorId 1)
[main] Start of transaction.
[MO] warning (Connector.cpp:534): tx process still running. Please call endTransaction(...) before
[main] Start of transaction.
[MO] warning (Connector.cpp:534): tx process still running. Please call endTransaction(...) before
[main] Start of transaction.
[MO] warning (Connector.cpp:534): tx process still running. Please call endTransaction(...) before
[main] Start of transaction.
[MO] warning (Connector.cpp:534): tx process still running. Please call endTransaction(...) before
[main] Start of transaction.
[MO] warning (Connector.cpp:534): tx process still running. Please call endTransaction(...) before
[main] Start of transaction.
[MO] warning (Connector.cpp:534): tx process still running. Please call endTransaction(...) before
```

matth-x, 2023-12-28:
Hi @pedro-fuoco, thanks for the appreciation, and welcome again to this open source project!

This indeed looks weird. I would first investigate the transaction lifecycle, maybe there's an explanation.

It could be related to the timing of the tx-related functions. `isTransactionRunning()` indicates if the lib has sent the StartTransaction request, but not yet the StopTransaction request. As `beginTransaction()` may not immediately send a StartTransaction request (e.g. because it waits for the connector plug), `isTransactionRunning()` may still be false after successfully creating the transaction via `beginTransaction()`. Here's an example sequence:

- Idle state
- after calling `beginTransaction()`: enter Beginning state. Now `isTransactionActive()` is true and `getTransactionIdTag()` returns the idTag. `isTransactionRunning()` is still false
- lib has sent StartTx request: enter Running state. Now `isTransactionRunning()` is also true
- after calling `endTransaction()`: enter Ending state. Now `isTransactionActive()` is false, but `isTransactionRunning()` could still be true
- lib has sent StopTx request: revert to Idle state again.

The reason of separating begin- and startTx is to better support offline scenarios. The charger can initiate a transaction locally and already start charging while being offline. The function `ocppPermitsCharge()` returns true if a local transaction was successful and the OCPP configs allow charging then.

pedro-fuoco, 2024-01-02:
Solved it!
The issue occurred in the freeVend. Turns out I was using the configuration `FreeVendEnabled` (mimicking ABB's configuration key) instead of `MO_CONFIG_EXT_PREFIX "FreeVendActive"`. Correcting this and setting `setConnectorPluggedInput()` solved the problem.

closed 2024-01-02 by pedro-fuoco