matth-x/MicroOcpp #245: Problem building for ESP-IDF.

KasperHauge2, 2023-12-29:
Hello. 
I'm trying to get the MicroOcpp into my ESP-IDF platform using the examples/ESP-IDF. But I get a lot of errors. My main issue now is that I'm not sure how to configure the mongoose-idf implementation, since this seems to start form about ver. 7.8, and the description on the MicroOcpp require version 6.14 (MO_MG_VERSION_614), which are not supported by the mongoose-idf. 
I have follow the description in the example, but there are some mismatch between the folder names and the description, Can you make it clear for me which "components" I need to install for the ESP-IDF, and which defines I need to build it?  
Thanks !
BR
Kasper

matth-x, 2024-01-03:
Hi @KasperHauge2,

The Mongoose integration supports both MG v7.8 and v6.14. For new projects I would recommend using the newer version of course. Then the build flag MO_MG_VERSION_614 should not be set.

In case you've already tried this, can you post the compiler error? I must admit that the build check for the esp-idf doesn't work perfectly yet and the head commits of the main / master branch could indeed be incompatible with each other. In any case, the newest commit from MicroOcpp, cesanta/mongoose-esp-idf and ArduinoJson should work with MicroOcppMongoose at commit 92ebea9fe85.

Apart from the dependencies as specified in the [CI step for the esp-idf](https://github.com/matth-x/MicroOcpp/blob/master/.github/workflows/esp-idf.yml) or the [example folder structure](https://github.com/matth-x/MicroOcpp/tree/master/examples/ESP-IDF/components), there are no further components to install.

KasperHauge2, 2024-01-14:
Thanks Matth. 
When running the example code, and change the ArduinoJson to v6.21.5 and the MicroOcppMongoose to v1.0.0, then I can build it, and I think it works !

closed 2024-01-14 by KasperHauge2

matth-x, 2024-01-14:
@KasperHauge2,

Thank you for bringing this issue up. It's necessary to pin the ArduinoJson version to 6 now and I changed the docs accordingly.

commit 12d48682 (2024-01-14): pin ArduinoJson to v6.x (fixes #245)