matth-x/MicroOcpp #253: StatusNotification info field

martamartz, 2024-01-24:
Hello, good morning!

How can I add the optional field "info" in StatusNotification.req when I execute ocpp_setConnectorPluggedInput_m??
Currently, with ocpp_setConnectorPluggedInput_m, I manage to transition the connector to Preparing correctly, but I also want the "info" to be sent in that Status.

I am implementing code in C.
Thank you very much in advance!

matth-x, 2024-01-24:
Hi @martamartz,

Unfortunately, it is not possible to fill the info field with the ConnectorPluggedInput. You may specify an error code and info if you set the ErrorDataInput (C++):

https://github.com/matth-x/MicroOcpp/blob/12d4868200d9675d98ff81b9c226e5deace64c12/src/MicroOcpp.h#L273

To add such custom functionality, it's necessary to go beyond the MicroOcpp_c.h facade and interact with the C++ modules. That's not too complex in C. Add two extra files "ocpp_helper.h" and "ocpp_helper.cpp" to your repository. ocpp_helper.cpp will contain the C++ code and interact with the internals of the OCPP lib. ocpp_helper.h contains C-style functions inside these linkage specifiers:

https://github.com/matth-x/MicroOcpp/blob/12d4868200d9675d98ff81b9c226e5deace64c12/src/MicroOcpp_c.h#L46-L48

https://github.com/matth-x/MicroOcpp/blob/12d4868200d9675d98ff81b9c226e5deace64c12/src/MicroOcpp_c.h#L188-L190

Then your C app can call the functions declared in ocpp_helper.h which will execute the C++ code in ocpp_helper.cpp (there are great tutorials out there for more details).

In ocpp_helper.cpp, you can register an ErrorCodeInput specifying the info field for an error StatusNotification message.

If it's not important that the info comes with the connector plugged StatusNotification, you may also send a custom StatusNotification message. Add a function like this:

```cpp
void ocpp_statusNotification(const char *info) {
    auto context = MicroOcpp::getOcppContext();
    //create a StatusNotification object and send it via `context`
}
```

Creating a StatusNotification object looks like this:

https://github.com/matth-x/MicroOcpp/blob/12d4868200d9675d98ff81b9c226e5deace64c12/src/MicroOcpp/Model/Authorization/AuthorizationService.cpp#L179-L192

Instead of setting `cpStatus` for connector 0, you would have to pass the status of the current connector. Replace "LocalListConflict" with an [ErrorData](https://github.com/matth-x/MicroOcpp/blob/master/src/MicroOcpp/Model/ConnectorBase/ChargePointErrorData.h) object.

martamartz, 2024-02-05:
Hello again,
I have managed to send a StatusNotification with the info field.
However, when doing it without ocpp_setConnectorPluggedInput, I think the library doesn't know that the EV is connected, and therefore, when I want to charge, a Start is not sent.

How could I solve this?

matth-x, 2024-02-05:
I would restore the old ocpp_setConnectorPluggedInput and send the custom StatusNotification messages on top of that. That adds a little redundancy but I would be surprised if it breaks the server.

closed 2024-02-06 by martamartz