matth-x/MicroOcpp #255: How to add configuration key

demirschmid61, 2024-01-30:
Hello,

I am using one variable (setPoint), I want to give access to CMS users to change that setPoint variable. Need to make the setPoint variable as the configuration key.

Can you please tell me how to make the setPoint variable as the configurationKey?

My expectation is, the CMS user should be able to change that configurationKey (setPoint).

Is it possible?

matth-x, 2024-02-04:
Hi @demirschmid61,

Yes, it is possible. You can declare custom configs which the library will handle like any other configuration then. Here is the how-to: https://github.com/matth-x/MicroOcpp/issues/248#issuecomment-1883585103

demirschmid61, 2024-02-06:
@matth-x ,

Thank you very much, I'll just implement it in the same way.

closed 2024-02-06 by demirschmid61