matth-x/MicroOcpp #257: Corruption of filesystem with blackouts when porting to ESP32

imanolmc, 2024-02-04:
Hello,

I am encountering an issue with blackouts affecting the filesystem. Whenever a blackout occurs, the filesystem becomes corrupted. Consequently, the system fails to restart because it cannot locate the necessary files to recover past transactions. I have experimented with both Spiffs and LittleFS, but encountered similar issues with both filesystems. Additionally, attempts to unregister the required partition during blackouts have not resolved the problem.

Do you have any insights on how to address this issue? Alternatively, how does MicroOCPP handle blackouts?

Thank you for your assistance.

Best regards,

matth-x, 2024-02-04:
Hi @imanolmc,

I don't have this issue on my devices (and actually it's the first time I see that LittleFS error message). My first guess is that maybe the partition table points to odd places on the flash, my second guess is that the flash is actually damaged. Or something goes wrong when mounting the file system? (the built-in integrations for Arduino and ESP-IDF SPIFFS provide an option to mount the partition on the host file system. When writing a custom fs integration, that's something which has to be done in the custom code)

Sorry that I have no further idea for the moment which part in the MO sources could explain this issue. If you find anything suspicious though, I would appreciate if you share it!

imanolmc, 2024-03-27:
Hello Matth,

I am going to close the issue because the problem was a bug with the ESP processor.

Thanks for your help!

Imanol

closed 2024-03-27 by imanolmc