matth-x/MicroOcpp #258: (BootNotification.cpp:83): request has been Rejected

FlxMojojojo, 2024-02-07:
I'm working on the charger, in which the server is developed for Python OCPP. the device is connected to the URL, but the boot notification is rejected. is there any change is standard with these two? I'm adding logs for reference.
[image]
serial monitor log: 
info (Connection.cpp:69): Connected to url: ******************************* // url replaced  with *
17:57:49.806 -> [MO] info (BootNotification.cpp:83): request has been Rejected
17:58:49.870 -> [MO] info (BootNotification.cpp:83): request has been Rejected
17:59:49.914 -> [MO] info (BootNotification.cpp:83): request has been Rejected

FlxMojojojo, 2024-02-10:
@matth-x When Micro ocpp tries to connect to - https://github.com/mobilityhouse/ocpp  server.  There is problem with the URL. 

The path which is suppose to have only the device ID, is also getting in the URL any idea why ?

Baruch055, 2024-03-28:
I am having a similar problem, with an ESP32 controller.
The error looks like this:

~~~
[MO] info (MicroOcpp.cpp:296): initialized MicroOcpp v1.0.2
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (Connection.cpp:65): Disconnected
[MO] info (RequestQueue.cpp:45): operation timeout: BootNotification
[MO] info (Connection.cpp:65): Disconnected
~~~
I tested using [MicroOcppSimulator](https://github.com/matth-x/MicroOcppSimulator),
And it works fine against my server.
So the problem is not with the server, apparently.

Baruch055, 2024-03-30:
@FlxMojojojo 
I had the same problem, and it was solved like this:
I changed to `wss` instead of `ws`.
Try it.

Charles-Isaac, 2024-04-29:
Seems to be related to #278 which has been fixed in #283

closed 2024-05-18 by matth-x