matth-x/MicroOcpp #259: Operation timeout while resending statusNotification when websocket reconnects

demirschmid61, 2024-02-10:
@matth-x 

I upgraded the library from ArduinoOcpp to MicroOcpp and resending the status notification when websocket disconnects. I am facing an issue of **"operation timeout: StatusNotification"** while resending status notifications. The wifi network connection given to charger is also stable, am I missing anything while resending the status?

**My code snippet is as follows.**

```
    for (size_t connectorId = 0; connectorId < numberOfConnector; connectorId++)
    {
        auto statusNotification = makeRequest(new StatusNotification(connectorId, 
                                                                           context->getModel().getConnector(connectorId)->getStatus(),
                                                                           context->getModel().getClock().now(), nullptr));

        statusNotification->setTimeout(60000);
        context->initiateRequest(std::move(statusNotification));
    }  
```

**The logs are**

[MO] info (StatusNotification.cpp:50): New status: Available (connectorId 0)
[MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): StopTransaction
[MO] info (StatusNotification.cpp:50): New status: Preparing (connectorId 1)
[MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): StatusNotification
[MO] info (StatusNotification.cpp:50): New status: Preparing (connectorId 2)
**[MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): StatusNotification**
[MO] Recv: WS pong

**[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification**
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] Recv: WS pong

[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] Recv: WS pong

[MO] Recv: WS ping **(Websocket already connected)**
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] Recv: WS pong

Can you please share your input on the above issue?

Also please suggest your inputs for the server side on the following scenario.

Step 1: Charger boot up, websocket connected
Step 2: Boot Notification sent
Step 3: Status Notification sent
Step 4: Working fine for next 10 min
**Step 5: WebSocket disconnected**
after 1/2(variable) minute
Step 6: WebSocket reconnect
**My question at this point, do I need to resend the status notification according to OCPP standard?**

matth-x, 2024-02-17:
Hi @demirschmid61, that looks fine in my opinion. OCPP relies on the TCP retransmission mechanism for lost packages, but on top of that, the standard doesn't define a general strategy for delivery guarantee (apart from the `TransactionMessageAttempts` for transaction-related messages). When the TCP connection is reset, the StatusNotifiction operation is lost.

It could be indeed a nice improvement to initiate a new StatusNotification when the charger gets reconnected. Redundant StatusNotfications do no harm. To check if the WebSocket has reconnected, observe the return value of `unsigned long Connection::getLastConnected()`. If it changes, then it means that a new connection has been established.

demirschmid61, 2024-02-21:
@matth-x 
Thank you for your input, I'll check and let you know.

closed 2024-02-26 by demirschmid61