matth-x/MicroOcpp #261: Firmware service is not sending all FirmwareStatusNotification

demirschmid61, 2024-02-27:
Thank you very much @matth-x for this wonderful library, 

I followed the steps for HTTPS firmware update, its working. The file is downloading and installing perfectly, but the CP is not able to send the following FirmwareStatusNotification,

1. **Downloading**
2. **Downloaded**

I can see only the installing status first and installed after the charger reboots and found new build.

Do I need to edit the state machine to send FirmwareStatusNotification?

```
[MO] Recv: [2,"2055a53f-1a2a-45c8-b968-fd0a62b61e2e","UpdateFirmware",{"location":"https://raw.githubusercontent.com/abcd/sdscsdc/firmware.bin","retrieveDate":"2024-02-22T17:11:06.260Z"}]
[MO] info (FirmwareService.cpp:211): Scheduled FW update!
                  location = https://raw.githubusercontent.com/esp32update/main/firmware.bin
                  retrieveDate = 2024-02-22T17:11:06.260Z
                  retries = 1, retryInterval = 180
[MO] Send: [3,"2055a53f-1a2a-45c8-b968-fd0a62b61e2e",{}]
[MO] info (FirmwareService.cpp:62): Start update
[MO] debug (Connector.cpp:391): Status changed
[MO] info (StatusNotification.cpp:50): New status: Unavailable (connectorId 0)
[MO] Send: [2,"1000010","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Unavailable","timestamp":"2024-02-22T17:11:06.941Z"}]
[MO] Recv: [3,"1000010",{}]
[MO] Send: [2,"1000011","FirmwareStatusNotification",{"status":"Installing"}]
[MO] Recv: [3,"1000011",{}]
```

priya2212, 2024-03-02:
Hi @demirschmid61 , Did you get any solutions for your problem? I am also facing the same issue. Would be happy if you can share your inputs. :)

demirschmid61, 2024-03-04:
@priya2212 , I'll be happy to share, but the problem is not yet solved.

aditya111222, 2024-06-13:
@priya2212 @demirschmid61 how we can add the firmware update in esp-idf?