matth-x/MicroOcpp #273: BeginTransaction process with 2 connectors

martamartz, 2024-03-27:
Hello Matth,

I'm working with a charger with two connectors (connectorId = 1 and connectorId = 2). When a card is detected and both connectors are available, I would like to start charging on the connectorId where the car is plugged in (taking into account ocpp_setConnectorPluggedInput_m and ConnectionTimeOut).

To achieve this, I have tried:
ocpp_beginTransaction_m(1, idTag); and ocpp_beginTransaction_m(2, idTag); but this sends two Authorize messages when only one is needed.

I have also tried:
ocpp_beginTransaction_m(0, idTag); and then plug into connectorId 2. It only sends one Authorize, but the charging does not start on 2.

Is there any way to do this? Or is the only way for the library to default to assigning it to connectorId 1?

Thankss!!

imanolmc, 2024-04-17:
Dear Matth,

I am having more or less the same issue as the previous message. I am using an RFID card without selecting any connector in the charger. The process starts but it is always assigned to the connectorID 1, is there any possibility to start the transaction without a connector and including the connectorID after detecting that the car is plugged to one of the connectors.

Regards,

jonlarra11, 2024-04-23:
Hey Matth,

I'm having trouble starting a charge after detecting an RFID. Upon accepting the card, without having a specific connector assigned, it always chooses one of the two, regardless of whether there is a connected EV in the other connector.

 I was wondering if there was a way for me to choose which connector is plugged in.

Thx in advance!

martamartz, 2024-05-15:
Having a generic begin_transaction for connector 0 would be great, Matth! 
Do you have any estimate of when this could be done? 

Thank you very much

matth-x, 2024-05-21:
It will take me a couple of months to work on it, there are quite a few things stapling up on my bucket list. Probably it's best to solve this on your end by manually sending an Authorize messag and finally calling beginTransaction.

closed 2024-07-04 by martamartz