matth-x/MicroOcpp #278: Bug in void mocpp_initialize(const char *backendUrl,...

samuellavoie90, 2024-04-11:
In mocpp_initialize(const char *backendUrl,...  a websocket is initialized:

webSocket->begin(host.c_str(), port, url.c_str(), "ocpp1.6");

The entire url (ws://Server.local:8500/ocpp/ChargerId) is passed to this websocket, when this library (arduinoWebSockets) was actually expecting only to have the path (/ocpp/ChargerId) passed there, this results in a badly formed HTTP packet for the initial websocket creation handshake.

[image]
In the screenshot you can sort of see the intended use is for URL to only have the Path component, since the default value is '/'

matth-x, 2024-04-13:
Hi @samuellavoie90, thanks for this report!

Did you already try to change the url parameter of webSocket->begin() accordingly? Did it work?

matth-x, 2024-04-18:
Thanks for taking a deeper look, will fix it like suggested.

Actually this confused me for years. It _is_ possible to send the full URL in HTTP requests, but it's really uncommon and quite a few WS servers had issues with it. I thought the WS lib had a weird behavior there, but it turns out it was just me 😅

matth-x, 2024-04-21:
@samuellavoie90 The fix is ready now: https://github.com/matth-x/MicroOcpp/pull/283

Can you check if it works with your WS server?

closed 2024-05-18 by matth-x