matth-x/MicroOcpp #288: Issue with Inconsistent Responses for StopTransaction Requests

AbdullahJalloul, 2024-04-30:
Hi @matth-x .
I am currently implementing your library in a project and have encountered an intermittent issue that I hope you can help me with. Occasionally, when sending a StopTransaction request from the server to the charging station, the station receives and executes the command, but the response returned to the server is empty. This does not happen all the time, but it is frequent enough to cause concern.

[image]

Could you please advise on what might be causing these empty responses? Are there known issues with message handling or network reliability that I should be aware of? Any insights or suggestions you could provide would be greatly appreciated.

Thank you for your support and for the great work on the library.

Best regards,

matth-x, 2024-05-01:
Hi @AbdullahJalloul,

Thank you for the report. Does the charger send meter values within the StopTransaciton, i.e. is something set up in the config keys StopTxnSampledData or StopTxnAlignedData? My first guess is that the payload JSON overflows, but this is only plausible if the transactionData has a regression bug.

AbdullahJalloul22, 2024-05-02:
Hi @matth-x,

Thank you for your response. I checked the configurations and noticed that both StopTxnSampledData and StopTxnAlignedData are indeed populated. The current setup includes parameters like Power.Active.Import, Energy.Active.Import.Register, Voltage, and Current.Import.

Here's how they are declared in the configuration:

```
declareConfiguration<const char*>("MeterValuesSampledData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
    declareConfiguration<const char*>("StopTxnSampledData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
    declareConfiguration<const char*>("MeterValuesAlignedData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
    declareConfiguration<const char*>("StopTxnAlignedData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
```

Given your suggestion about a possible JSON payload overflow, could these settings be contributing to the issue? If so, what adjustments would you recommend to mitigate this?

Your insights are greatly appreciated.
Best regards,

AbdullahJalloul, 2024-05-02:
Hi @matth-x,

Thank you for your response. I checked the configurations and noticed that both StopTxnSampledData and StopTxnAlignedData are indeed populated. The current setup includes parameters like Power.Active.Import, Energy.Active.Import.Register, Voltage, and Current.Import.

Here's how they are declared in the configuration:

```
declareConfiguration<const char*>("MeterValuesSampledData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
    declareConfiguration<const char*>("StopTxnSampledData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
    declareConfiguration<const char*>("MeterValuesAlignedData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
    declareConfiguration<const char*>("StopTxnAlignedData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
```

Given your suggestion about a possible JSON payload overflow, could these settings be contributing to the issue? If so, what adjustments would you recommend to mitigate this?

Your insights are greatly appreciated.
Best regards,

AbdullahJalloul, 2024-05-02:
Thank you for your quick response. Yes, I am indeed using StopTxnSampledData and StopTxnAlignedData configurations. Here's how they are set up:

```
declareConfiguration<const char*>("MeterValuesSampledData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
declareConfiguration<const char*>("StopTxnSampledData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
declareConfiguration<const char*>("MeterValuesAlignedData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
declareConfiguration<const char*>("StopTxnAlignedData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import");
```

As you can see, I am sampling and aligning multiple data types like Power, Energy, Voltage, and Current at the end of transactions. Your suggestion about the possibility of JSON payload overflow makes sense, considering the amount of data involved. Could you please advise on how to confirm if this is the case and any potential workarounds or adjustments I might consider to mitigate this issue?

Thank you once again for your support and guidance.

Best regards,

matth-x, 2024-05-13:
Thanks for the information. I tried to reproduce this using the Simulator on Linux, but without luck. I will troubleshoot it on a microcontroller platform too.

Another possible scenario is that the device is running low on heap memory and then the JSON serialization fails. Do you monitor the heap usage and could there be a connection with the blank message? In any case, the library is lacking a meaningful error message if the allocation fails here which needs to be improved in future.