matth-x/MicroOcpp #293: Meter values in ESP-IDF

aditya111222, 2024-05-05:
How can I add my metering value ?

martamartz, 2024-05-13:
You can declare a function like this:
float SetPower(){
    return your_instantaneous_power;
}

And in setup, after
OCPP_Connection *osock = ocpp_makeConnection(&mgr,
            EXAMPLE_MO_OCPP_BACKEND, 
            EXAMPLE_MO_CHARGEBOXID, 
            EXAMPLE_MO_AUTHORIZATIONKEY, "", fsopt);
    ocpp_initialize(osock, "ESP-IDF charger", "Your brand name here", fsopt, false);

add:
ocpp_addMeterValueInputFloat(SetPower,"Power.Active.Import", "W", NULL, NULL);  

Just call ocpp_addMeterValueInputFloat once here. The library will take the values of your_instantaneous_power as needed according to MeterValueSampleInterval

aditya111222, 2024-06-10:
Close

closed 2024-06-10 by aditya111222