matth-x/MicroOcpp #299: Saving the charging summary

chandan-ultraviolette, 2024-05-15:
I want to save the charging summary, basically one .jsn should be created at every new charging session and periodically gets updated while charging. I think something similar is already done with s-*.jsn file, I enabled below but the stack is not creating the file sd-*.jsn. Any suggestion.

[image]

[image]

[image]

matth-x, 2024-05-18:
Are you sure the value of StopTxnSampledData was applied correcly? You can verify it with a GetConfiguration command from the server. Otherwise, I would believe that the microcontroller runs out of flash memory and therefore, the creation of the StopData JSON files fails. I verified this function here and couldn't reproduce the issue.

matth-x, 2024-05-18:
By default, the number of StopData measurements is limited to 4 which could be too restrictive in your scenario. I quickly changed the corresponding definition. To keep more than 4 elements on flash, set the build flag `MO_MAX_STOPTXDATA_LEN` to a higher value.

chandan-ultraviolette, 2024-05-21:
@matth-x  I am bit confusion, I thought after 4 elements, code will erase the older ones and keep creating new log?

chandan-ultraviolette, 2024-05-21:
@matth-x I recompiled my entire code working fine now, seems very strange. Also I don't want to create new file periodically, basically one file sd-connector_id-txNr-0 should update periodically so I can access only 1 file to know the history.

Also I checked the JSON sd-*.jsn, I was looking for JSON which has start time, end time (or duaration) and energy consumed wh till now, Arre you creating any such .jsn or any suggestion where to modify for same.

matth-x, 2024-05-21:
The current strategy is to make a consecutive record of up to `MO_MAX_STOPTXDATA_LEN -1` sd files and then to keep the most recent readings in the last sd file which gets updated afterwards. So the sd file with the highest sequence number always contains the most recent measurements.

chandan-ultraviolette, 2024-05-22:
Understood, How can I access txNr so application can access the logs.

matth-x, 2024-05-22:
The transaction store keeps track of all currently stored transactions. You can retrieve the range of corresponding txNr like that:

```cpp
if (auto context = getOcppContext()) {
    auto& model = context->getModel();

    if (auto txStore = model.getTransactionStore()) {
        int txNrBegin = txStore->getTxBegin(1 /* connectorId */); //first possible txNr
        int txNrEnd= txStore->getTxBegin(1 /* connectorId */); //last possible txNr + 1

        if (txNrEnd == txNrBegin) {
            //tx history is empty
        } else {
            //all txNr from txNrBegin to txNrEnd - 1 are valid
        }
    }
}

```

chandan-ultraviolette, 2024-05-31:
Thanks @matth-x , One doubt as post reboot if last state was charging, it continues the charging. I think mocpp also storing the energy consumed wh so how to fetch that number from mocpp stack so can start increasing the energy consumption from that number.

matth-x, 2024-06-14:
Hi @chandan-ultraviolette, MicroOcpp does not terminate running transactions automatically when being initialized. This needs to be explicitly done by calling `endTransaction()` after `mocpp_initialize()` (not required to terminate it, but most chargers do it). Although the Start-/StopTx energy values are being stored by MicroOcpp once the operation has been initiated, that's not the case for the EnergyMeterInput over reboots. In general, the MeterValues, Start- and StopTx operations will always capture the original value of the EnergyMeterInput, there is no processing / correction step in between. It's the responsibility of the firmware to persist the energy register over reboots.

StopTxnSampledData should contain the measurands, like "Energy.Active.Import.Register".