matth-x/MicroOcpp #300: Function to define the current limit of a EVSE

p-r-m-n, 2024-05-17:
Is there any function in the C API of this library that allows to change the current limit of the EVSE? I understood that "ocpp_setSmartChargingCurrentOutput" uses a function to tell when the limit has changed by the car, but what about the limit of the EVSE itself?

matth-x, 2024-05-18:
In the standard Smart Charging scenario, the OCPP server sends charging limits to the EVSE which MicroOcpp forwards to your application via the `ocpp_setSmartChargingCurrentOutput` callback function.

The EVSE can report back the actual charging limit (resulting from local load balancing restrictions, cable capacity, ect.) over MeterValues with the Measurand `Current.Offered`. To add a new measurand, use

https://github.com/matth-x/MicroOcpp/blob/c0732dccacd13a7b2a091272b0eaf92be26a5c30/src/MicroOcpp_c.h#L137

For example:

```c
float my_charging_limit() {
    return 16.f; // TODO replace with the actual charging limit of the EVSE
}

// register callback (somewhere in the setup function): 
ocpp_addMeterValueInputFloat(my_charging_limit, "Current.Offered", "A", NULL, NULL);
```

p-r-m-n, 2024-05-18:
I understood this callback, but what i am trying to do is set the maximum current that the evse can supply the vehicle with. In the Schneider evlink park EVSE, this setting can be changed through ocpp, as shown in the picture (Page 32 of the evlink comissioning guide).

https://github.com/matth-x/MicroOcpp/blob/c0732dccacd13a7b2a091272b0eaf92be26a5c30/src/MicroOcpp_c.h#L137

matth-x, 2024-05-21:
I see. It's possible to register custom Configurations as described here: https://github.com/matth-x/MicroOcpp/issues/248#issuecomment-1883585103 The approach works the same for accessing built-in Configs as for declaring new, custom Configs.

Having a Config for the maximum current, you can read its value in the firmware integration and make sure that your charger never exceeds that level manually.

p-r-m-n, 2024-05-22:
> I see. It's possible to register custom Configurations as described here: [#248 (comment)](https://github.com/matth-x/MicroOcpp/issues/248#issuecomment-1883585103) The approach works the same for accessing built-in Configs as for declaring new, custom Configs.
> 
> Having a Config for the maximum current, you can read its value in the firmware integration and make sure that your charger never exceeds that level manually.

thanks, i will try and post the results here.

p-r-m-n, 2024-06-09:
I tried to connect a esp to a EVSE but apparently i had it wrong, this library should connect to a ocpp server as the EVSE does, not the other way. There are any functions in this library to implement a ocpp server on the esp32?

matth-x, 2024-06-10:
No, it's not supported at the moment. I did build a proof of concept for this once, but didn't go any further that direction.

closed 2024-06-10 by matth-x

p-r-m-n, 2024-06-10:
thanks for the support!

> No, it's not supported at the moment. I did build a proof of concept for this once, but didn't go any further that direction.