matth-x/MicroOcpp #305: Commercially Available Options

hasanr68, 2024-05-21:
Hi.

Unsure where to ask questions - please let me know if there's a google group account to ask questions. I am looking for a commercially available solution with the functionality of this library - has anyone implemented it already? If yes, possible to share link? If not, is there a complete install guide with hardware list to make it work? I am specifically looking to switch on/off electric sockets once payment is made via QR Code, card etc. and has to be compatible with OCPP.

Thanks in advance and apologies!

matth-x, 2024-05-21:
I'm planning to publish a simple OCPP socket implementation very soon. It will support payment via QR code, however, RFID-based authorization won't be supported in the beginning. I will post an update in this issue when it's open sourced.

hasanr68, 2024-05-22:
Hi Matt.
Thanks a lot for the quick response. That sounds great. Any idea about when it will happen? Also - will it be OCPP compliant, since the sockets wont have Control Pilot (CP) wire?

matth-x, 2024-06-03:
Hi @hasanr68, the OCPP Smart Socket is released! See: https://github.com/matth-x/OCPP-Socket

For sure, OCPP was mainly designed for J1772-type connectors. However, in some countries like France it's also quite common that charge points provide a simple socket for household socket chargers. These chargers don't use all features of OCPP, but they can be operated with this protocol very well.

closed 2024-06-03 by matth-x

hasanr68, 2024-06-04:
Thanks a lot! :)