matth-x/MicroOcpp #307: FreeVending Mode for Two conectors Simultaneously

demirschmid61, 2024-05-23:
Hello @matth-x,

I am implementing the free vending mode for two connectors.

Successfully implemented free vending mode for 1 connector, the following configuration keys are mentioned below.

**Cst_FreeVendActive** set to **true**
**Cst_FreeVendIdTag** set to **ABCDEFG**

I also added the idTag **ABCDEFG** to the CMS server.

Problem statement:
If I want to use the free vend mode for both the connectors, can I?
I also checked the configuration key generated in the file **/ocpp-config.jsn**, and only one idTag **Cst_FreeVendIdTag** key is available.

Is it possible to **simultaneously** charge the vehicle in free vending for both the connector and with the single **Cst_FreeVendIdTag**?

Do I need to do any other configuration?

matth-x, 2024-05-27:
Hi @demirschmid61,

To answer your questions,

1. Yes, I would say using the same idTag for both connectors is an acceptable solution. I have never seen a backend system which requires unique idTags per connector. To test it with SteVe, it's necessary to increase the maximum concurrent transactions for the freevend idTag.
2. If you set Cst_PreBootTransaction to true, then no OCPP connection is necessary in freevend mode. However, the transaction queue will get full at some point and by default, MicroOcpp will reject further offline transactions. To override that behavior, set Cst_SilentOfflineTransactions to true, then the next transactions will become "silent" when the queue is full and won't occupy any space.

demirschmid61, 2024-05-29:
@matth-x Thank you very much for your reply.
I'll implement this and let you know the outcome.

demirschmid61, 2024-06-05:
@matth-x 

The following keys are configured for the charger.
Cst_PreBootTransactions = True
Cst_SilentOfflineTransactions = True
Cst_FreeVendIdTag = "abcde"

**My observation:**
1. The charging transaction is perfectly executed and only recent 4 transactions are saved in the flash

**Problem Faced:**
**1. After some (e.g.: 12) transactions, if somehow the charger gets reset (Manually or remotely), the charger is not starting the transaction in free vending mode, provided the charger is not connected to the internet**

**Please have a look at the logs free vending working perfectly:**
`
------------------------**For start transaction in free vending**------------------------------------------
14:17:19.870 > [MO] info (Connector.cpp:430): begin FreeVend Tx using idTag **abcde**
14:17:20.562 > [MO] info (StatusNotification.cpp:50): New status: Preparing (connectorId 1)
14:17:20.731 > [MO] info (Connector.cpp:290): Session mngt: trigger StartTransaction
14:17:20.940 > [MO] info (StartTransaction.cpp:48): StartTransaction initiated
14:17:21.002 > [MO] info (StatusNotification.cpp:50): New status: Charging (connectorId 1)

------------------------**For stop transaction in free vending**------------------------------------------
14:19:52.250 > [MO] info (Connector.cpp:353): Session mngt: trigger StopTransaction
14:19:52.675 > [MO] info (StopTransaction.cpp:50): StopTransaction initiated
14:19:52.683 > [MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): StartTransaction
14:19:52.707 > [MO] info (StatusNotification.cpp:50): New status: Finishing (connectorId 1)
14:19:52.713 > [MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): StatusNotification
14:20:01.306 > [MO] info (StatusNotification.cpp:50): New status: Available (connectorId 1)
14:20:01.314 > [MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): MeterValues
14:20:02.211 > [MO] info (main.cpp:1994): MeterValues reset for connector 1
`

When the charger gets reset the freevending transactions does not started.

Please acknowledge on my observation, is it correct or not.

Am I missing something in the configuration keys?

matth-x, 2024-06-14:
Hi @demirschmid61, this could actually be related to a regression bug in v1.0. Can you upgrade to v1.1?

Everything on your side looks correct. Cst_PreBootTransactions and Cst_SilentOfflineTransactions are both set correctly and Freevend is also properly set up.

demirschmid61, 2024-06-15:
@matth-x Thank you very much,

The **free vending mode** for **both connectors** perfectly worked for me and regarding pure offline free vending mode, I'll upgrade the library version to 1.1 and let you know the observations.

demirschmid61, 2024-08-14:
@matth-x Thank you for your input,

I just upgraded to 1.1 and the free vending transactions are working perfectly for both connectors even if the charger gets rebooted.

My last objective for free vend mode is not to connect my charger to the OCPP server **ever** and to use the charger in free vend mode. Is it possible or at least once i need to connect it with the server for bootNotification?