matth-x/MicroOcpp #308: cppcheck fail (cppcheck --force src/MicroOcpp)

mundel, 2024-05-24:
src/MicroOcpp/Operations/GetConfiguration.cpp:138:13: error: There is an unknown macro here somewhere. Configuration is required. If MO_DBG_DEBUG is a macro then please configure it. [unknownMacro]
            MO_DBG_DEBUG("Unknown key: %s", key)

matth-x, 2024-05-27:
Hi @mundel, thank you for the hint! I recently changed the debug macros and there could indeed be an issue with the new macros definition. I would like to reproduce this. How did you use cppcheck to find that error? Btw. I would be very open to adding cppcheck as a GitHub Action.

mundel, 2024-05-27:
Hi @matth-x ,
As I mentioned in heading.
Use below command to check:-
cppcheck --force src/MicroOcpp

matth-x, 2024-05-27:
Couldn't reproduce it here. The output for GetConfiguration (on main) is

````
Checking src/MicroOcpp/Operations/GetConfiguration.cpp ...
67/92 files checked 82% done
``

Any idea what I could have missed?

mundel, 2024-05-31:
Hi @matth-x 
I was cross check and found with your latest patch this issue is not seen.
I was not sync code from last couple of days.

mundel, 2024-05-31:
issue not seen with latest code.

closed 2024-05-31 by mundel