matth-x/MicroOcpp #311: First StatusNotification 1970-01-01 timestamp

martamartz, 2024-06-04:
Hello everyone,

I have configured the EVSE with Cst_PreBootTransactions set to true to allow transactions before it connects to Steve. I have discovered that with this configuration, the initial StatusNotifications are sent with the timestamp 1970-01-01, despite having received and set currentTime in the BootNotification confirmation.

I assume this is related to issue #230, but I don't know how to fix it to have the correct date.

Thank you very much in advance!

[OCPP] info (MicroOcppMongooseClient.cpp:492): connection ws://url_to_steve-- connected!
[OCPP] Send: [2,"1000000","BootNotification","chargePointModel":"a","chargePointVendor":"a","firmwareVersion":"a","chargePointSerialNumber":"1112345678","meterSerialNumber":"a","meterType":"a","chargeBoxSerialNumber":"1112345678","imsi":"123","iccid":"123"}]
[OCPP] Recv: [3,"1000000",{"status":"Accepted","currentTime":"2024-06-04T12:37:40.090Z","interval":120}]
[OCPP] info (BootNotification.cpp:100): request has been Accepted
[OCPP] Send: [2,"1000001","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"1970-01-01T00:00:00.740Z"}]
[OCPP] Send: [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"1970-01-01T00:00:00.750Z"}]
[OCPP] Send: [2,"1000003","StatusNotification",{"connectorId":2,"errorCode":"NoError","status":"Available","timestamp":"1970-01-01T00:00:00.760Z"}]

matth-x, 2024-06-14:
Hi @martamartz, good finding! Let's keep this issue open until I've found a solution to this.

As a workaround, you could set the time manually right after `mocpp_initialize()`. Then the StatusNotification messages will use that value. Even if you don't have an accurate time, it's maybe better for the backend system to be only few hours off than having dates in 1970. The BootNotification will synchronize the OCPP time again. Here's how to set the time manually: https://github.com/matth-x/MicroOcpp/issues/24#issuecomment-1881579330

matth-x, 2024-06-15:
Checked it again, MicroOcpp will only initiate StatusNotification messages if the clock has been set (except for reporting errors which isn't the case here). Tested it again and couldn't reproduce your issue. Could it be that you're using an older version of MicroOcpp?

matth-x, 2024-06-15:
While revisiting the StatusNotification messages I did find a scenario where MicroOcpp would send a message with timestamp 1970-01-01. Error codes which occur before BootNotification were potentially sent without a valid timestamp. Glad you reported your finding.