matth-x/MicroOcpp #315: Unable to Remote Start Transaction strangely one day

chandan-ultraviolette, 2024-06-11:
I am using OCPP 1.6J stack. All working fine but one day I started the setup and sent the RemoteStartTransaction command but stack was rejecting it. Later after debug, I found out connector->getTransaction() was returning the non zero value due to same, stack was rejecting the request and printing "No connector to start transaction". When I request for connector status, connector was available with no error.

I restarted the system but again issue. May be transaction data was stored in non volatile memory and loaded back, any suggestion on solving this ?

matth-x, 2024-06-15:
I believe this is due to a regression bug which has already been resolved. Did you already upgrade to v1.1?

Nevertheless, the connector would still report to be Available in similar circumstances (while the connector is blocked by a stale transaction). I changed this behavior so that it sends Finishing, to make similar malicious behavior easier to track down.

chandan-ultraviolette, 2024-06-15:
@matth-x  I found the bug. Basically, after stopping the charging, I was accessing the tx-connector_id-txNo.jsn file and uploading it to the cloud. During this time, the OCPP stack was updating the complete field in the tx-connector_id-txNo.jsn file after receiving confirmation from the OCPP CMS. As a result, it was unable to update. Therefore, once I restarted the system, the OCPP stack still considered the last session as not over.

I have to handle this issue in firmware. 

Also I didn't update to v1.1 as I didn't lots of changes in OCPP stack based on my firmware requirement, can you suggest which file you modified for the changes related to same? So I can manually modify it.