matth-x/MicroOcpp #316: Server error, data loss!

meylence, 2024-06-11:
Hi @matth-x ,

I am trying to set up an electric charging station system for my home, but I have been encountering errors for a week, as shown in the images below. Could you help me understand the possible reasons for these errors?

[image]

[image]

matth-x, 2024-06-15:
Hi @meylence,

The StopTransaction operation fails and MicroOcpp drops it, so the corresponding data is likely lost.

The server reports the error message "Invalid ID Tag" so my first guesses are:

- MicroOcpp does not send the idTag after RemoteStopTransaction, but the server requires it (if you're willing to share which backend you're using, I would be interested!)
- The idTag is not properly configured in the backend system

Maybe the server logs provide more information.

meylence, 2024-06-28:
When I look at the backend software, I encounter an interesting phenomenon.  When I choose normal shutdown in the backend software, everything works normally and it shuts down the system.
[image]

However, When I shut down with the force shut down option, the system crashes here.
[image]