matth-x/MicroOcpp #319: ESP32-IDF ESP32C3 : How to  test all the ocpp  opetrations over esp32 module

Kundanjhakj, 2024-06-13:
Hi Developers,
I am working with the ESP32-C3 module and need to test all operations such as BootNotification, Heartbeat, StatusNotification, etc., with CMS. I have built MicroOCPP code on the Espressif IDE. Additionally, I have successfully connected it with CMS. Could anyone please guide me on how to send each operation one by one for testing purposes?

Thank you.

matth-x, 2024-06-15:
Hi @Kundanjhakj,

There is no systematic approach of making MicroOcpp send all operations one by another. For example, MicroOcpp will send the FirmwareStatusNotification only as part of a firmware update process, but it has no API for sending it manually.

Still you can rebuild each operation using the CustomOperation class like this

https://github.com/matth-x/MicroOcpp/blob/1bf49421a4641a29ddaf5e3de99027bd93d2b22a/tests/Reservation.cpp#L91-L100

However, I'm not sure if that is exactly what you mean.

Kundanjhakj, 2024-06-17:
Hi, In this code, esp32 uses wifi to connect with websocket. I have to connect with 4G module

matth-x, 2024-06-20:
You can replace the built-in WebSocket client by your own WebSocket client which you have full control of. Then you can instruct the WebSocket client to route all traffic over 4G. The custom WebSocket client needs to subclass the MicroOcpp `Connection` interface and implement all its functions. To make MicroOcpp use the custom WS client, you need to pass it to `mocpp_initialize` like this: https://github.com/OpenEVSE/openevse_esp32_firmware/blob/ff671ff90cfbc8ec794bd4e7481301758b867934/src/ocpp.cpp#L243-L255

There's also a user who has used the TinyGSM project with MicroOcpp. See this issue for a demonstration: https://github.com/matth-x/MicroOcpp/issues/238#issuecomment-1877293070

Kundanjhakj, 2024-09-03:
Hi, I am having an issue.
I have created another thread for meter value reading.

c
Copy code
int getEnergyReading(void) {
    return energy;
}

static void energy_reading(void *arg) {
    while (1) {
        ocpp_setEnergyMeterInput(getEnergyReading);
        vTaskDelay(pdMS_TO_TICKS(100000)); // 60-second delay
    }
}
The issue is that when I run it, it gives the same value multiple times on the CMS side.
please see log file.

[log.txt](https://github.com/user-attachments/files/16844329/log.txt)

ap9389552, 2024-10-19:
@Kundanjhakj Did you tested all the operation over esp32? if yes can you guide me

Kundanjhakj, 2024-10-21:
Hello @ap9389552, yes its working.

ap9389552, 2024-10-21:
Actually Im working on esp-idf framework And Im unable to get the meter reading ,can we chat on personal please here is my mail id :
akashpra699@gmail.com