matth-x/MicroOcpp #320: GetDiagnostics‘s issue

shengxiang200788, 2024-06-13:
Hi Matth-x，
First, a lot thanks to your great work.  
Nowdays,I was testing the lib,  and got items for you:
1. getDiagnostics function
In file src/MicroOcpp/Operations/GetDiagnostics.cpp, about line 40,

Timestamp stopTime;
    if (payload.containsKey("startTime")) {
        if (!stopTime.setTime(payload["stopTime"] | "Invalid")) {
            errorCode = "PropertyConstraintViolation";
            MO_DBG_WARN("bad time format");
            return;
        }
    }
seems the if condition is not correct.

2. still getDiagnostics function
Another issue is when payload has no stopTime and startTime,  these two are not initialized and passed to requestDiagnosticsUpload(...) 

3.  also getDiagnostics function
There is a callback refreshFilename(), it gives  user  a chance to define the uploading file name,
I want to define a filename with more details like starttime-stoptime.zip or something else to let 
the user get details about the uploaded file.  So it may seems better to add some parameters to refreshFilename() .

matth-x, 2024-06-14:
Hi @shengxiang200788,

Thanks a lot for the report!

1. Fixed it on main.
2. The default constructor initializes it to 1970-01-01 which means "undefined". The callee should check the timestamp values first:

```cpp
if (startTime >= MIN_TIME) {
    //startTime defined, process it here
}
```
3. Nice idea, will think about it. As a workaround for now, you can get the original payload of the GetDiagnostics by setting an "onRequest" listener which is executed with the original OCPP message payload:

https://github.com/matth-x/MicroOcpp/blob/47cc0e4b853b193ac9154d4a481759afd196ae29/src/MicroOcpp/Core/OperationRegistry.h#L35

Then you could store the startTime and stopTime strings somewhere and read them again when `refreshFilename()` is called. Setting the onRequest listener won't interfere the normal execution in the DiagnosticsService.

matth-x, 2024-06-15:
Sorry, need to row back on this. The onRequest listener is executed after the internal MicroOcpp handlers, and in this case only after `refreshFilename()` gets called.

If I get more feedback about the start- and stopTime in the diagnostics filename, I will add these fields to the `refreshFilename()` signature as well. For now, the only valid solution is to "rebuild" the GetDiagnostics operation by copying GetDiagnostics.h and GetDiagnostics.cpp into your own sources, changing conflicting names and then overloading the internal GetDiagnostics similar to this:

https://github.com/matth-x/MicroOcpp/blob/47cc0e4b853b193ac9154d4a481759afd196ae29/src/MicroOcpp/Model/Diagnostics/DiagnosticsService.cpp#L27-L28

The overload needs to happen after the initialization of DiagnosticsService.