matth-x/MicroOcpp #321: Offline Start transaction using local list

chandanbhatia, 2024-06-14:
I want to start Transaction during offline time so I set the local list from CMS. When I am starting session using ocpp_beginTransaction_authorized_m(2, "tag", NULL); or ocpp_beginTransaction_m(2, "tag"), gets waring "tx process still running. Please call endTransaction(...) before" but as soon as able to connect with  CMS, there is no such issue. Please let me know how to start the offline session.

chandanbhatia, 2024-06-14:
I debugged more and found out that connector loop doesn't run post reboot until connected to CMS, that'y Transaction is not send to NULL.

Also please confirm if ocpp_beginTransaction_authorized_m(2, "tag", NULL) is right method to start offline charging session.

matth-x, 2024-06-15:
By default, MicroOcpp doesn't start any transactions before the first BootNotification succeeds. That's because the BootNotification could reject the charger, so technically the charger needed to wait for the result. However, it's very common to bypass that behavior. To do so with MicroOcpp, set the configuration Cst_PreBootTransactions to true. "PreBoot" refers to the phase before the BootNotification succeeds.

`ocpp_beginTransaction_authorized_m(idTag)` will overwrite the built-in offline behavior, bypass local lists and assume that `idTag` has already been authorized somewhere else. `ocpp_beginTransaction_m(idTag)` will query the local list, try to send an Authorize message and handle the offline case like defined by AllowOfflineTxForUnknownId.