matth-x/MicroOcpp #322: Ethernet (W5500) Support with Mongoose WebSocket Adapter on Arduino platform

demirschmid61, 2024-06-15:
Hello @matth-x 
 
Thank you very much for the library and WebSocket adapter, the library is helping me for the development of AC charger.

I am developing an OCPP client with ethernet and using **Arduino framework**. When I use **MicroOcppMongoose** with wifi it's perfectly working, but when trying to connect using ethernet, ESP32  crashes.

**---------Logs when connected using ethernet----------**
Local IP : 192.168.0.238   
Subnet Mask : 255.255.255.0
Gateway IP : 192.168.0.2   
DNS Server : 192.168.0.14        
Ethernet Successfully Initialized

[MO] warning (MicroOcppMongooseClient.cpp:202): Insecure connection (WS)
assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace: 0x400836a5:0x3ffb1f10 0x4008a231:0x3ffb1f30 0x4008f679:0x3ffb1f50 0x401084d5:0x3ffb2080 0x401161ad:0x3ffb20b0 0x4011620d:0x3ffb20d0 0x401081d1:0x3ffb2120 0x400f504c:0x3ffb2140 0x400f4bb6:0x3ffb2160 0x400f8261:0x3ffb2180 0x400f837f:0x3ffb22e0 0x400f844e:0x3ffb2370 0x400f89ab:0x3ffb2410 0x400f91e6:0x3ffb2530 0x400d2ed9:0x3ffb25c0 0x400fba26:0x3ffb2820

  #0  0x400836a5:0x3ffb1f10 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x4008a231:0x3ffb1f30 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x4008f679:0x3ffb1f50 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x401084d5:0x3ffb2080 in tcpip_send_msg_wait_sem at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:455 (discriminator 1)
  #4  0x401161ad:0x3ffb20b0 in netconn_apimsg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:136
  #5  0x4011620d:0x3ffb20d0 in netconn_new_with_proto_and_callback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:166
  #6  0x401081d1:0x3ffb2120 in lwip_socket at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/sockets.c:1774
  #7  0x400f504c:0x3ffb2140 in socket at C:/Users/atspc/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/sockets.h:656
      (inlined by) mg_socket_if_connect_tcp at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:3864
  #8  0x400f4bb6:0x3ffb2160 in mg_do_connect at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:3168
  #9  0x400f8261:0x3ffb2180 in mg_connect_opt at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:3345
  #10 0x400f837f:0x3ffb22e0 in mg_connect_http_base at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:8784
  #11 0x400f844e:0x3ffb2370 in mg_connect_ws_opt at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:10449
  #12 0x400f89ab:0x3ffb2410 in MicroOcpp::MOcppMongooseClient::maintainWsConn() at .pio/libdeps/esp32-development-board/MicroOcppMongoose/src/MicroOcppMongooseClient.cpp:219 (discriminator 4)
  #13 0x400f91e6:0x3ffb2530 in MicroOcpp::MOcppMongooseClient::MOcppMongooseClient(mg_mgr*, char const*, char const*, char const*, char const*, std::shared_ptr<MicroOcpp::FilesystemAdapter>, MicroOcpp::ProtocolVersion) at .pio/libdeps/esp32-development-board/MicroOcppMongoose/src/MicroOcppMongooseClient.cpp:76
  #14 0x400d2ed9:0x3ffb25c0 in setup() at src/main.cpp:175
  #15 0x400fba26:0x3ffb2820 in loopTask(void*) at C:/Users/atspc/.platformio/packages/framework-arduinoespressif32@3.20006.221224/cores/esp32/main.cpp:42
**------------------------------------------------------------------------------------------**

**-------------------------------Logs when connected using wifi with **MicroOcppMongoose** adapter-----------------------**
[main] Wait for WiFi: ........[MO] 
warning (MicroOcppMongooseClient.cpp:202): Insecure connection (WS)
[MO] info (MicroOcpp.cpp:349): initialized MicroOcpp v1.1.0
[MO] info (MicroOcppMongooseClient.cpp:474): connection ws://192.168.1.22:1884/fu5v1 -- connected!

[MO] Send: [2,"1000000","BootNotification",{"chargePointModel":"AC","chargePointSerialNumber":"GN234400","chargePointVendor":"myComp","firmwareVersion":"0.0.10"}]
[MO] Recv: [3,"1000000",{"currentTime":"2024-06-11T04:41:41.831724900Z","interval":900,"status":"Accepted"}]
[MO] info (BootNotification.cpp:93): request has been Accepted
[MO] info (StatusNotification.cpp:53): New status: Available (connectorId 0)
[MO] info (StatusNotification.cpp:53): New status: Available (connectorId 1)
[MO] info (StatusNotification.cpp:53): New status: Available (connectorId 2)
[MO] Send: [2,"1000001","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-06-11T04:41:41.843Z"}]
[MO] Recv: [3,"1000001",{}]
[MO] Send: [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2024-06-11T04:41:41.844Z"}]
[MO] Recv: [3,"1000002",{}]
[MO] Send: [2,"1000003","StatusNotification",{"connectorId":2,"errorCode":"NoError","status":"Available","timestamp":"2024-06-11T04:41:41.855Z"}]
[MO] Recv: [3,"1000003",{}]
**------------------------------------------------------------------------------------------**

**My Questions**
1. Is the Mongoose WebSocket adapter perfectly initialized?
2.  Am i missing any build flag setting in pltform.ini file

#--------------------------Main.cpp--------------------------#
`
#include <Arduino.h>
#include <WiFi.h>
#include <MicroOcpp.h>
#include "mongoose.h"
#include <MongooseCore.h>
#include <MicroOcpp/Core/Connection.h> //need for setting TLS credentials
#include <MicroOcpp/Core/Configuration.h>
[... 130 lines cut ...]
lib_deps = ${common.lib_deps}
monitor_speed = ${common.monitor_speed}
build_flags =
    -D MG_ENABLE_SSL=1
    -D MG_ENABLE_HTTP_STREAMING_MULTIPART=1
    -D MG_ENABLE_EXTRA_ERRORS_DESC=1
    -D MG_SSL_MBED_DUMMY_RANDOM=1
    -D MG_SSL_IF=MG_SSL_IF_MBEDTLS
    -D MG_SSL_IF_MBEDTLS_FREE_CERTS=1
    -D MG_ENABLE_SNTP=1
    -D CS_PLATFORM=CS_P_ESP32
    -D MO_MG_VERSION_614 ; use MG v6.14
    -D ENABLE_WIRED_ETHERNET
    -D MO_CUSTOM_WS
    -D MO_DBG_LEVEL=MO_DL_INFO    ; Uncomment to see OCPP info logs
	-D MO_CUSTOM_DIAGNOSTICS     	; Uncomment to see MO_CUSTOM_DIAGNOSTICS
	-DMO_TRAFFIC_OUT
    -D MO_NUMCONNECTORS=3
    -D ARDUINOJSON_ENABLE_STD_STRING=1

board_build.partitions = min_spiffs.csv
upload_speed = 921600
monitor_filters =
    esp32_exception_decoder`
#-----------------------------------------------------------------#

Parthacngreen, 2024-06-17:
Error compiling for board DOIT ESP32 DEVKIT V1

matth-x, 2024-06-20:
I can remember that I also dealt with that "Invalid mbox" crash. The solution was to delay the Mongoose initialization until getting the IP address from the Arduino framework. It was a plain `while (!ipConfigured){delay(1000);}` loop before the MicroOcpp stuff in the `setup()` function.