matth-x/MicroOcpp #324: How to send 'Finishing' when receive 'SoftReset' during charging. Not 'Unableable'

JungHeum-Park, 2024-06-19:
I'm testing scenario of receive `SoftReset` when charging.
To pass OCTT test, EVSE should send `StopTransaction` and `StatusNotification` with `Finishing` status after receive `SoftReset`.
But when receive `SoftReset`, internal system automatically set statue to `Unavailable`.

Where should I modify in the library to get the results I want?

```
[MO] Send: [2,"1000007","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2024-06-19T09:28:48.479Z"}]
[MO] Recv: [3, "1000007", {}]
[MO] Recv: [2, "067d6516-bcf0-41f6-beb7-91d9d338fbdc", "Reset", {"type":"Soft"}]
[MO] Send: [3,"067d6516-bcf0-41f6-beb7-91d9d338fbdc",{"status":"Accepted"}]
[MO] info (Connector.cpp:333): Session mngt: trigger StopTransaction
[MO] info (StopTransaction.cpp:51): StopTransaction initiated
[MO] Send: [2,"1000008","StopTransaction",{"idTag":"d9104c52","meterStop":21,"timestamp":"2024-06-19T09:28:59.053Z","transactionId":1718783682,"reason":"SoftReset"}]
[MO] info (StatusNotification.cpp:52): New status: Unavailable (connectorId 0)
[MO] info (StatusNotification.cpp:52): New status: Unavailable (connectorId 1)
[MO] Recv: [3, "1000008", {"idTagInfo":{"status":"Accepted"}}]
[MO] info (StopTransaction.cpp:204): Request has been accepted!
[MO] Send: [2,"1000009","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Unavailable","timestamp":"2024-06-19T09:28:59.393Z"}]
[MO] Recv: [3, "1000009", {}]
```

matth-x, 2024-06-19:
Hi @JungHeum-Park,

Thanks for reporting this! There is a fix on the way already for that OCTT test case. If you can patch the MicroOcpp sources in your build, comment that line out and it should be fine:

https://github.com/matth-x/MicroOcpp/blob/1bf49421a4641a29ddaf5e3de99027bd93d2b22a/src/MicroOcpp/Model/Reset/ResetService.cpp#L104

JungHeum-Park, 2024-06-21:
Now I pass the test.
thanks to @matth-x