matth-x/MicroOcpp #327: How to make specific feature not support

JungHeum-Park, 2024-06-21:
Hi.

When server request `SupportedFeatureProfiles` of `GetConfiguration`, I want the device response without including `LocalAuthListManagement`.

Where can I config the supported feature?

matth-x, 2024-06-21:
Hi @JungHeum-Park, you can set the build flag `MO_ENABLE_LOCAL_AUTH=0` which disables that feature and the corresponding entry in the SupportedFeatureProfiles won't show up.

closed 2024-06-21 by JungHeum-Park

razvanphp, 2025-10-20:
Unfortunately the OCTT is really exaggerating on this one, so the test case fails:

```
Started testcase LocalAuthListManagement_CS - TC_043_1_CS-Send Local Authorization List - NotSupported
[...]
2025-10-20 16:31:31.239
msg-out
[
  2,
  "f90acc26-d4fd-4b35-bbe2-e1ed0607bae1",
  "SendLocalList",
  {
    "listVersion": 1,
    "localAuthorizationList": [
      {
        "idTag": "17E004D9",
        "idTagInfo": {
          "status": "Accepted"
        }
      }
    ],
    "updateType": "Full"
  }
]
2025-10-20 16:31:31.240
info
Waiting for response message of type SendLocalListResponse
2025-10-20 16:31:31.240
info
Waiting for response message of type SendLocalListResponseReceive or CallError.
2025-10-20 16:31:31.427
msg-in
[
  4,
  "f90acc26-d4fd-4b35-bbe2-e1ed0607bae1",
  "NotImplemented",
  "",
  {}
]
2025-10-20 16:31:31.440
info
CallError.ErrorCode should be 'NotSupported'
2025-10-20 16:31:31.441
info
The value of 'Reset CS after testcase' is false
```

I wrote a ticket to support, maybe they will add `NotImplemented` as well as valid response, makes no sense to add a dummy Operation just for this, especially when you want to disable those features for performance reasons.