matth-x/MicroOcpp #328: configuration key 'ChargingScheduleAllowedChargingRateUnit' accessibility is wrong

JungHeum-Park, 2024-06-21:
Hi.
When server request `GetConfiguration`

When the server sent the request `GetConfiguration`, `"readonly"` of 'ChargingScheduleAllowedChargingRateUnit' in the response 'configurationKey' was incorrect.
According to the ocpp1.6 document, the accessibility of 'ChargingScheduleAllowedChargingRateUnit' is R(read only), but the library returns '"readonly": false'.

JungHeum-Park, 2024-06-21:
I find how to change accessibility.

Add `true` as last parameter.
https://github.com/matth-x/MicroOcpp/blob/1bf49421a4641a29ddaf5e3de99027bd93d2b22a/src/MicroOcpp/Model/SmartCharging/SmartChargingService.cpp#L321

matth-x, 2024-06-21:
Thank you for the report!

commit 8b1f0fdb (2024-06-21): fix ChargingScheduleAllowedChargingRateUnit mutability (fixes #328)
closed 2024-06-21 by matth-x