matth-x/MicroOcpp #332: Scheduled charging on Power cut

Aylaas, 2024-07-01:
Hi @matth-x,

I have a test case scenario where there is an existing schedule for a charger, and power goes off.
In this case, I expect my charger to retain the schedule profiles and start charging on power on, if the scheduled profile still holds good(schedule duration is valid and within the bounds).
But, while testing this case, I don't see the charger starting to charge on its own on the power on and the schedule still active.

Can you please provide feedback on this test case, if valid please provide the insights on the same.

Thanks,
Ayla.

matth-x, 2024-07-01:
Hi @Aylaas, by default, MicroOcpp stores all charging schedules locally and resumes transactions after a power loss. Is there any hint that the flash access doesn't work properly?

Aylaas, 2024-07-17:
Hi @matth-x , 
After Power loss, only the latest schedule is retained. 
Suppose I set two profiles, one with "ChargePointMaxProfile" the next one with "TxDefaultProfile", there is power loss, on next power regain, only "TxDefaultProfile" is retained and the charge profile count is 1.

How do I retain all the schedules and read back on next power cycle.

matth-x, 2024-07-28:
This implementation normally keeps all schedules. I assume that something with the flash memory goes wrong, for example the flash goes out of memory. In this case MicroOCPP would show error messages on the log. Any suspicious messages there?