matth-x/MicroOcpp #333: Plugged connector check

Wert137-dsp, 2024-07-02:
Hello Matth.

I want to check when the charger is plugged into the vehicle, there is the setConnectorPluggedInput function, but I didn't find anything within the lib to retrieve this information, I wrote a code to resolve this issue but I would like to know if there is already a solution.

Wert137-dsp, 2024-07-02:
> Hello Matth.
> 
> I want to check when the charger is plugged into the vehicle, there is the setConnectorPluggedInput function, but I didn't find anything within the lib to retrieve this information, I wrote a code to resolve this issue but I would like to know if there is already a solution.

Wert137-dsp, 2024-07-02:
.

matth-x, 2024-07-05:
Hi @Wert137-dsp, there is no method to read back those values. When I work on firmware and need the connector status somewhere else, I usually store the status in a separate variable which I can fully access before reporting it to MicroOcpp.

Wert137-dsp, 2024-07-11:
Hi @matth-x , do you have any example code for implementing this part that you can provide?