matth-x/MicroOcpp #336: Idleness check after end-transaction

Wert137-dsp, 2024-07-12:
hello @matth-x 

I'm working on an implementation to check idleness after the end of loading, I would like to know if there is anything ready about this, thank you very much.

matth-x, 2024-07-12:
By idle you mean that the charger has finished the transaction and is Available again? You can read the OCPP status here:

https://github.com/matth-x/MicroOcpp/blob/8b1f0fdbae8785c49453f0b2d54a1a8db2fbe3ed/src/MicroOcpp.h#L248-L251

E.g.

```cpp
if (getChargePointStatus() == ChargePointStatus_Available) {
    // Charger is idle
}
```

Wert137-dsp, 2024-07-16:
@matth-x 

idle in the sense that the user has finished charging and kept the plug connected.

It seems that the Context class is able to take the values of the loading states, including finalization.

matth-x, 2024-07-28:
One possibility is to read the OCPP ChargePointStatus and check if it is in state Finished:

https://github.com/matth-x/MicroOcpp/blob/af66b0a5ca2ec2eee2316993eb22f7e2c0d26a6d/src/MicroOcpp.h#L251

```cpp
if (getChargePointStatus() == ChargePointStatus_Finished) {
    // The transaction has stopped, but the EV is still plugged
}
```

The ChargePointStatus prioritizes Faulted, Unavailable or Reserved status over Finishing though, so that approach could not suffice your use case. The only alternative is to implement the logic on your own, i.e. track if there is a transaction going, track if the EV is plugged and in case the transaction is dismissed but EV remains plugged, you detect the "idle" state.