matth-x/MicroOcpp #338: Linux 下cmake编译工程出现 "major" is defined冲突？如何解决？

JUMP-ZYX, 2024-07-17:
[build] /home/MicroOcpp/./src/MicroOcpp/Version.h:30:13: warning: In the GNU C Library, "major" is defined
[build]  by <sys/sysmacros.h>. For historical compatibility, it is
[build]  currently defined by <sys/types.h> as well, but we plan to
[build]  remove this soon. To use "major", include <sys/sysmacros.h>
[build]  directly. If you did not intend to use a system-defined macro
[build]  "major", you should undefine it after including <sys/types.h>.
[build]      ProtocolVersion(int major = 1, int minor = 6, int patch = 0) : major(major), minor(minor), patch(patch) { }

‘const int MicroOcpp::ProtocolVersion::minor’ should be initialized     

出错代码
/*
 * OCPP version type, defined in Model
 */
struct ProtocolVersion {
    const int major, minor, patch;
    ProtocolVersion(int major = 1, int minor = 6, int patch = 0) : major(major), minor(minor), patch(patch) { }
};

}

ryanchang419, 2024-09-10:
Me same ! do you solve this problem ?

nithinmanne, 2024-09-27:
I was able to get the code to build by making the following change in `Version.h`. But I'm not sure what the cause is, or why this change fixes it.

Add this before definition of `struct ProtocolVersion`.
```
#ifdef major
#undef major
#endif

#ifdef minor
#undef minor
#endif
```

Git Patch:
```
diff --git a/src/MicroOcpp/Version.h b/src/MicroOcpp/Version.h
index 92a2ea4..a61fa19 100644
--- a/src/MicroOcpp/Version.h
+++ b/src/MicroOcpp/Version.h
@@ -22,6 +22,14 @@
 
 namespace MicroOcpp {
 
+#ifdef major
+#undef major
+#endif
+
+#ifdef minor
+#undef minor
+#endif
+
 /*
  * OCPP version type, defined in Model
  */
```