matth-x/MicroOcpp #346: Charging Profil with Kind TxProfil is rejected without TransactionId

MGeigl, 2024-08-01:
I have send a setChargingProfil with ProfilKindType TxProfil during an active transaction but i always received a rejected.

SetChargingProfile {"connectorId":1,"csChargingProfiles":{"chargingProfileId":5,"stackLevel":1,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-01T07:00:00.000Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6,"numberPhases":3}]}}}]

I guess the problem is that we do not send a transaction id, which is optional corresponding to the standard (7.8 Charging Profile - transactionId - integer - 0..1 - Optional. Only valid if ChargingProfilePurpose is set to TxProfile, the transactionId MAY be used to match the profile to a specific transaction.)

But at https://github.com/matth-x/MicroOcpp/blob/a2c6307b48bf4ba311e22f56d7175d3156dc74c3/src/MicroOcpp/Operations/SetChargingProfile.cpp#L68 the id needs to be set to accept the profil.

matth-x, 2024-08-02:
Hi Michael, thanks for this report! I fixed it in #348, can you check if it solves your issue?

MGeigl, 2024-08-19:
Hi Matthias
as far as i can remeber, the loadChargingProfil function returned 0 as transaction Id. Therefore, i think >= 0 will fail again. 
https://github.com/matth-x/MicroOcpp/blob/a2c6307b48bf4ba311e22f56d7175d3156dc74c3/src/MicroOcpp/Operations/SetChargingProfile.cpp#L40

Is TransactionID 0 even valid as ID? I thought it needs to be at least 1. Then i would change your >= 0 to >0.

Unfortunatly, i am not able to test it this week.

MGeigl, 2024-08-22:
ok, forget my last comment. the loadChargingProfile delivers -1 if no transactionId is included and therefore your version with >=0 is correct and works. I tested it just now. 
so from my side it is ready to merge.

commit 4ea7db2d (2024-10-08): make txId in TxProfile an optional parameter (fixes #346) (#348)
closed 2024-10-08 by matth-x