matth-x/MicroOcpp #396: Integrating MicroOCPP on STM32F446RE Without Built-in Wi-Fi

kiranj26, 2025-01-19:
Hello, and thank you for this fantastic library! I am currently working on integrating MicroOCPP on an STM32F446RE Nucleo board. However, the **STM32F446RE** does not have built-in Wi-Fi capabilities like the ESP32. To provide networking functionality, I plan to use an **ESP32-C3 development board** as a Wi-Fi interface.

## **Context**
I am working on integrating **MicroOCPP** on an **STM32F446RE Nucleo board**, which lacks built-in Wi-Fi capabilities. To enable networking, I plan to use an **ESP32-C3** as a Wi-Fi interface. The STM32F446RE will handle the MicroOCPP stack, WebSocket communication, and TLS encryption, while the ESP32-C3 will act as a simple Wi-Fi-to-UART bridge.

### **Plan**
1. **STM32F446RE**:
   - Runs the MicroOCPP library.
   - Uses **MbedTLS** for TLS and **Mongoose** for WebSocket communication.
   - Communicates with ESP32-C3 over UART.

2. **ESP32-C3**:
   - Acts as a Wi-Fi bridge.
   - Establishes Wi-Fi connection and forwards TCP/WebSocket data to/from the STM32.

### **Questions**
1. Are there challenges when porting MicroOCPP to STM32F4 without native networking?
2. Is using ESP32-C3 as a Wi-Fi-to-UART bridge a feasible approach, or do you suggest alternatives?
3. Are Mongoose and MbedTLS suitable for WebSocket and TLS on STM32F4, or would you recommend other tools?

### **Request**
Looking forward to your guidance or any suggestions to optimize this approach. Thank you for this fantastic library!

razvanphp, 2025-05-20:
Just curious, isn't STM32 supposed to be supported already? I saw STM mentioned on the README, hence the my confusion.

Also, maybe you want to check https://github.com/matth-x/MicroOcpp/issues/172#issuecomment-1669574485.

Aweisile69, 2025-06-17:
Hello, have you made any progress? 
I am doing something similar to you. However, I am not very familiar with C++, so I encountered a lot of difficulties. 
I use GD32F470VI as the main control, run the MicroOCPP library, and use the W5500 Ethernet chip (integrated hardware TCP/IP protocol stack) for network communication. 
Some time ago, I ported mongoose and used the websocket-related API in this library to simply test the connection of the ocpp server (sent bootnotification and got the correct response). 
At that time, I wrote a very simple program myself. 
Later, I found this MicroOCPP library, which is used by so many people and seems to have relatively complete functions, so I want to try to port it. 
If you have any progress, I hope you can reply to me, thank you!