matth-x/MicroOcpp #407: Connection to SteVe?

pinuea, 2025-03-28:
Hi! 
I have SteVe working and now I´ve seen MicroOcpp they are compatible? This is like a graphic layer of SteVe?
I enter here https://demo.micro-ocpp.com/ and it seems good.
I can install in my server and connect to SteVe? I will see my Charge Points?

I see in the two projects @juherr 

Thanks!

juherr, 2025-03-28:
MicroOcpp is a C++ library that implements the OCPP protocol for microcontrollers.

The URL you mentioned points to a station simulator based on MicroOcpp. Its source code is available here: https://github.com/matth-x/MicroOcppSimulator.

Steve is a Charging Station Management System (CSMS) that provides charging services to users and can manage many stations.

These two projects are compatible: a charging station running MicroOcpp will work seamlessly with Steve, as well as with many other CSMS platforms.

If you’re looking for an open-source CSMS, Steve is definitely worth considering.
If you’re looking for open-source station firmware, OpenEVSE is a great option: https://github.com/OpenEVSE/openevse_esp32_firmware.

pinuea, 2025-03-28:
Hi juherr,

This is what I understand. 
1-MicroOcpp is for microcontrollers. It is to install in a charge point. 
2-OpenEVSE is a open-source firmware, to instal in a charge point.
3-SteVe is a CSMS to manage charge points and idTags. I have this and I use it.
My charge points have OCPP and are connected to SteVe, so I think MicroOcpp and OpenEVSE are not for me.

But I would like to have more graphic options in SteVe? I thought MicroOcpp could solve it.
Any option?

thank you!

juherr, 2025-03-28:
> Any option?

Yes, you can either contribute directly to the Steve project or fork it to customize it for your needs.